jekhokie/raspberry-noaa-v2 #260: Enable all supported enhancements

colinluthier, 2021-03-12:
wx2img has many enhancements we don't yet support. Adding them is the kind of easy coding I can help with.

colinluthier, 2021-03-15:
I got a start on this today as well. I should do a PR by next weekend.

colinluthier, 2021-03-30:
Lol, it's past next weekend. This is a busy work project. Still on my list.

dom-robinson, 2021-09-29:
I think they are all available now @colinluthier ?

colinluthier, 2021-09-29:
In RN1 this was easy. Now we have to add a file for each enhancement. We are about at least 10 of the lesser used enhancements. It's just busy work to add the files.

dom-robinson, 2021-09-30:
Ah yes - i just pulled this list from the wxtoimg CLI docs:

ZA
MB
MD
BD
CC
EC
HE
HF
JF
JJ
LC
TA
WV
WV-old
NO
MCIR
MSA
MSA-precip
MSA-Anaglyph
HVC
HVCT
sea
therm
veg
anaglyph
canaglyph
class
histeq
contrast
invert
bw

and we are currently running these: ZA MCIR MCIR-precip MSA MSA-precip HVC-precip HVCT-precip HVC HVCT therm avi

So you are right - there are many more we could work in. As you say - a time soak, but not difficult.

OK - thats on the 'todo' list :)

ve3elb, 2021-09-30:
Yes I would like to see more added. But also adding two separate processing scripts. Once for local processing and one for push to web, exp discord. The user can choose which files to process and can then choose which one get push to the web. Same as in the windows gui version.

dom-robinson, 2021-10-08:
(I added CC HE HF MD BD MB JF JJ LC TA WV NO sea)

colinluthier, 2021-10-08:
Nice!

closed 2021-10-09 by dom-robinson