jekhokie/raspberry-noaa-v2 #263: Webpanel

s9z4dz, 2021-03-12:
After installing a new copy of V2, I'm unable to open the webpanel

jekhokie, 2021-03-12:
@s9z4dz what is configured in your `config/settings.yml` file for the following properties:

```
enable_non_tls: false
web_port: 80
enable_tls: true
web_tls_port: 443
```

By default, a fresh install will install the web interface as TLS-enabled on port 443, so you probably want to try visiting `https://<YOUR_PI_IP>/` unless you've changed the above properties appropriately? If you want to use the default port 80/non-TLS, switch this property to true `enable_non_tls: true` re-run your `./install_and_upgrade.sh` script.

s9z4dz, 2021-03-12:
Many thanks for the quick response, I change from false to true  then rerun and it's working.Blessings and keep up the great work

closed 2021-03-12 by jekhokie