jekhokie/raspberry-noaa-v2 #264: Annotation improvement

Krolykke, 2021-03-12:
**Please Describe the problem to be solved**

Too often you cant read your annotation overlayed text on your images becauuse it matches the image under, or you cover too much of your image if you use bacground in HTML in the annotation template.

Instead of overlaying, why not add it to the bottom of the picture?

jekhokie, 2021-03-12:
@Krolykke are you suggesting essentially expanding the height of the image to create a black "bar" at the top or bottom, into which the annotation would be displayed?

Krolykke, 2021-03-12:
Yep, and if you dont configure it it should not figure.

jekhokie, 2021-03-12:
Note to self - this is likely best implemented by adding an option `overlay_annotation` and default to `true` (current behavior). If the user changes this to `false`, we would need to add black bars for the annotation only when the user selects an `image_annotation_location` that is *NOT* `[West|Center|East]`.

jekhokie, 2021-03-17:
PR submitted - will let tomorrow's passes come through to see how this works (tested and appears to be working, but I want to see it in action before I merge). Hopefully I can merge tomorrow evening latest.

closed 2021-03-17 by jekhokie