jekhokie/raspberry-noaa-v2 #269: [Error] Duplicate pass entries in schedule

colinluthier, 2021-03-14:
When updating settings passes can get duplicated it the tle has changed by a second. 

[image]

jekhokie, 2021-03-14:
I've seen this happen a couple times on my end too, haven't figured out why/how other than NTP drift being an issue (although I think you had some super accurate stratum server on your end, right?)

colinluthier, 2021-03-15:
I should never be more than +- .002 out although it may be a little farther with your ntp file. I fix it w\once and a while and then a pull erases my changes. There is no way I can be .1 second out as long as the time server is running and I pull it from my PC every minute and get notified if it errors.

Krolykke, 2021-03-21:
I see it too, time syncing with dk.pool.ntp.org, can't figure out how add image of it here unfortunately.

jekhokie, 2021-03-24:
So I actually think this is, at its root, an issue with the TLE file and/or `predict` command - and here's my reasoning.

The conflict (seconds off from original) is not actually an issue in the `atq` job output, only in the display via the web interface. What this means is when the `predict` command is run against the TLE file (which is downloaded each time the schedule script is run), when the `pass_start` field is calculated, it's using milliseconds, which is then injected into the database and used as a primary key. If this value for `pass_start` (in ms) is off by even 1 ms, you end up with a duplicate record in the database (again, this is the primary key), and since the TLE files are being downloaded each time the schedule script is run, a slight variation in the TLE being updated at its source and/or a rounding error in the `predict` app can cause this to occur.

My easy fix for this is to basically add a parameter to the `schedule.sh` script which will check for the existence of an `--update-tle` switch - if this switch is present, the TLE will be refreshed, so this should *only* happen when the cron executes the schedule script (the user running the schedule script should never pass this switch lest risk duplicate passes). If the issue is in fact small updates happening to the TLE at its source (`www.celestrak.com`), this should fix the issue. However, if the issue persists, this means that `predict` is having a rounding/estimation issue for very close (in ms) passes, which we will need to address separately.

jekhokie, 2021-03-24:
I'm going to close this by merging a PR that handles the "only download TLE files when the cron runs the script" scenario - that's not saying it is *the* fix (there could still be a rounding error that the `predict` command is resulting in which is causing the duplicate), but it's one of the fixes and we'll see where it takes us.

closed 2021-03-24 by jekhokie