jekhokie/raspberry-noaa-v2 #270: Error, while updating

tomamdy, 2021-03-14:
TASK [common : install dependencies] *******************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Failed to update apt cache: "}
	to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP *********************************************************************
localhost                  : ok=14   changed=0    unreachable=0    failed=1   

Error:   Something failed with the install - please inspect the logs above

jekhokie, 2021-03-14:
This indicates something is wrong with your OS apt cache. Try doing `sudo apt-get update`.

tomamdy, 2021-03-14:
The command doesn't works at the first moment. After deleting captured sat pictures by webpanel there was more space and I could update. The result I see tomorrow. May be there is a solution to make more space bij scripts on the sdcard. Because I believe that that was the problem, but I am a hardware men.

jekhokie, 2021-03-14:
@tomamdy great news, let me know how it works out. Just FYI, there is a pruning option you can configure to manage space on your drive: https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/pruning.md

tomamdy, 2021-03-16:
I had forgotten to remove the wav files. That cost too much space,

closed 2021-03-16 by tomamdy