jekhokie/raspberry-noaa-v2 #271: [Error] No image output.

pineapplee85, 2021-03-14:
I was using the original raspberry noaa software but began getting what was clearly errors in the image. 
I put a bug request in and decided to try this software.
After a fresh install I am getting the same issue. I even tried another install but still the same.
After trying to get help on this discord I am putting in a request here.

The SDR is receiving fine. In fact, when I listen to an audio file I can hear a NOAA sat; but the 'metronome' sound seems faster than usual. When piping the audio into wxtoimg manually it does not decode either. This makes me think it may be an issue with sox? 

Nothing seems strange in the logs but feel free to have a look.

[image]

**Screenshots and/or logs**
[raspberry-noaa-v2-error140320.zip](https://github.com/jekhokie/raspberry-noaa-v2/files/6137172/raspberry-noaa-v2-error140320.zip)

**Support Output**
./support.sh: line 15: warning: command substitution: ignored null byte in input
=============================================
Details about environment
=============================================
Current date/time:  1615742284
Repo git hash:      ac134d390592f209a08f782a449348bdd938e78a
Raspberry Pi Model: Raspberry Pi 2 Model B Rev 1.1
Architecture:       Linux noaa-server 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 G                                                                                        MT 2020 armv7l GNU/Linux
Num CPUs:           4
---------------------------------------------
USB Device Map:
  Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fas                                                                                        t Ethernet Adapter
  Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 14.5G  0 disk
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 14.2G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       14657232 4253368   9775928  31% /
  devtmpfs          440244       0    440244   0% /dev
  tmpfs             473524       0    473524   0% /dev/shm
  tmpfs             473524    6324    467200   2% /run
  tmpfs               5120       4      5116   1% /run/lock
  tmpfs             473524       0    473524   0% /sys/fs/cgroup
  tmpfs             204800       0    204800   0% /var/ramfs
  /dev/mmcblk0p1    258095   46273    211823  18% /boot
  tmpfs              94704       0     94704   0% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:         947048 kB
  MemFree:          772116 kB
  MemAvailable:     835836 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00000001
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no

**Server information (please complete the following information):**
 - OS: Raspbian
 - Python version Unknown
 - SDR device Nooelec Nesdr SmarTee

**Additional context**
Problem randomly began on the original raspberry noaa software.

jekhokie, 2021-03-15:
Adding a note from the Discord chat - seemingly a hardware issue. If you can, try out the SDR on another Raspberry Pi, and/or another SDR if possible.

jekhokie, 2021-03-15:
@pineapplee85 just checking if you were able to run a test with different hardware/if that has resolved the suspected hardware issue?

pineapplee85, 2021-03-15:
@jekhokie been at work all day. Will give it a go when I get home and report back soon

pineapplee85, 2021-03-15:
Can confirm the SDR is functioning correctly. Both FM bands using my NOAA antenna and connected to my QO-100 station.
This must be a software related issue :(

pineapplee85, 2021-03-20:
Closed due to trying another install method. Will open up as and when if needed.

closed 2021-03-20 by pineapplee85