jekhokie/raspberry-noaa-v2 #283: Update passes at noon

colinluthier, 2021-03-15:
In order to see passes before bedtime scheduling passes at noon for +12 hr to +36 hr would be beneficial.

jekhokie, 2021-03-15:
So it sounds like there are actually 2 different features embedded in this if I'm reading it correctly?

1. The ability to set the time for the `schedule.sh` script to run.
2. The ability to specify the future window of scheduled passes.

Is that correct?

colinluthier, 2021-03-15:
I am not sure if it needs a setting. Just a change in timing of the schedule script. If it ran at noon and captured midnight to midnight it would be fine.

jekhokie, 2021-03-17:
Is there a strong desire to do noon-based scheduling? I'm honestly wondering if we keep it at midnight but do a 36 hour view so you get a day and a half?

closed 2021-03-31 by jekhokie