jekhokie/raspberry-noaa-v2 #286: Blank orbit.tle after pi reboots

vu2ati, 2021-03-16:
Whenever pi reboots, weather.txt file is getting cleared and hence no tle output causing error in scheduling job. Problem getting solved by running schedule.sh manually. This happens after updating to recent to latest version

jekhokie, 2021-03-16:
Note to self, this is due to the fact that the schedule script attempts to contact the TLE source endpoint (www.celestrak.com) before a DHCP lease is assigned to the device/an IP assigned. We will need to update the `schedule.sh` script to account for this.

thecivvie, 2021-03-16:
Edit the crontab to read:

@reboot sleep 5 && /home/pi/raspberry-noaa-v2/scripts/schedule.sh

I find the 5 second delay (Pi 4) is enough to allow for the network to come up. Maybe a suggestion to change the ansible script to add the delay, say 10 seconds

jekhokie, 2021-03-16:
@thecivvie I have a PR coming that is a bit more sophisticated than that - I found that it can actually take anywhere from 10-15 seconds (at least in my own tests). PR coming shortly.

closed 2021-03-16 by jekhokie