jekhokie/raspberry-noaa-v2 #291: Passes shown in web if even though meteor_m2_schedule: false

Krolykke, 2021-03-17:
**Describe the bug**
Previous scheduled passes is still shown in web if "meteor_m2_schedule: false" is applied after passes have been scheduled.

I suspect the same would count for noaa_XX_schedule: false

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Wed 17 Mar 2021 06:25:59 AM CET
Current date/time (ms):  1615958759
Repo git hash:           8bdbcf6c189d4bf9963c9c53082c632b345cd60e
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.4
Architecture:            Linux RaspiSat 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [912	Wed Mar 17 20:07:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210317-200759 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1616008079 994 54 Northbound E
  * [913	Wed Mar 17 21:49:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210317-214907 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1616014147 919 33 Northbound W
  * [911	Wed Mar 17 11:55:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210317-115540 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1615978540 902 26 Southbound W
  * [909	Wed Mar 17 18:41:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210317-184104 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1616002864 923 70 Northbound W
  * [910	Wed Mar 17 10:14:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210317-101444 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1615972484 987 73 Southbound E
  * [906	Wed Mar 17 07:13:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210317-071342 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1615961622 931 39 Southbound E
  * [908	Wed Mar 17 17:02:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210317-170218 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1615996938 860 26 Northbound E
  * [914	Wed Mar 17 07:26:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210317-072658 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1615962418 1005 88 Southbound W
  * [907	Wed Mar 17 08:53:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210317-085337 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1615967617 923 44 Southbound W
  * [915	Wed Mar 17 17:20:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210317-172014 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 1615998014 950 69 Northbound E
---------------------------------------------
USB Device Map:
  Bus 002 Device 002: ID 152d:0576 JMicron Technology Corp. / JMicron USA Technology Corp. 
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
  sda      8:0    0 223.6G  0 disk 
  ├─sda1   8:1    0   256M  0 part /boot
  └─sda2   8:2    0 223.3G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks     Used Available Use% Mounted on
  /dev/root      230488648 29655668 191416812  14% /
  devtmpfs         3899536        0   3899536   0% /dev
  tmpfs            4031632        0   4031632   0% /dev/shm
  tmpfs            4031632    25052   4006580   1% /run
  tmpfs               5120        4      5116   1% /run/lock
  tmpfs            4031632        0   4031632   0% /sys/fs/cgroup
  tmpfs            1024000        0   1024000   0% /var/ramfs
  /dev/sda1         258095    48676    209420  19% /boot
  tmpfs             806324        0    806324   0% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:        8063264 kB
  MemFree:         3062096 kB
  MemAvailable:    7680880 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

jekhokie, 2021-03-17:
@Krolykke just to confirm - after you changed the setting in your `config/settings.yml`, you did in fact run the `./install_and_upgrade.sh` script, correct? I just double-checked the `schedule.sh` script (which is triggered by the `install_and_upgrade.sh` script) and it does in fact honor the setting so long as it's set in your `~/.noaa-v2.conf` file (which would only be updated if you ran the install script after updating the setting).

Krolykke, 2021-03-17:
Allways! ;)

Krolykke, 2021-03-17:
pi@RaspiSat:~/raspberry-noaa-v2 $ nano ./config/settings.yml (setting meteor_m2_schedule: false)
pi@RaspiSat:~/raspberry-noaa-v2 $ ./install_and_upgrade.sh

-same thing, previous scheduled M2 passes remain in the webinterface.

jekhokie, 2021-03-17:
Note to self: might be as simple as adding a SQL statement to the schedule script right after the `atrm` loop removes all `at` jobs:

`delete from predict_passes where pass_start > CUR_DAY_START_EPOCH;`

jekhokie, 2021-03-25:
This will be solved as part of the fix for #317 - I will commit this fix later this evening.

closed 2021-03-25 by jekhokie