jekhokie/raspberry-noaa-v2 #292: Skip passes if max alt. not occur within a specified range

Krolykke, 2021-03-17:
**Please Describe the problem to be solved**
Skip passes if max alt. not occur within a specified range 

**(Optional): Suggest A Solution**
Add config entry per satellite like:

noaa15_capture_vector_min: 175
noaa15_capture_vector_max: 359,9

noaa18_capture_vector_min: 175
noaa18_capture_vector_max: 359,9

noaa19_capture_vector_min: 175
noaa19_capture_vector_max: 359,9

meteor_m2_capture_vector_min: 175
meteor_m2_capture_vector_max: 359,9

jekhokie, 2021-03-17:
Nice idea - and I'll default the range to `0-360` so all passes are scheduled by default/none are skipped.

dom-robinson, 2021-09-29:
I think this was resolved ages ago - closing for now, but happy to reopen if i am wrong!

closed 2021-09-29 by dom-robinson