jekhokie/raspberry-noaa-v2 #295: [Error] Upgrade from original raspberry noaa

pineapplee85, 2021-03-20:
**Describe the bug**
I have tried to upgrade from the original raspberry-noaa software but when logging into the web panel I get this error:

_
Warning: SQLite3::query(): Unable to prepare statement: 1, no such column: pass_start_azimuth in /var/www/wx-new/App/Models/Pass.php on line 20

Fatal error: Uncaught Error: Call to a member function fetchArray() on boolean in /var/www/wx-new/App/Models/Pass.php:25 Stack trace: #0 /var/www/wx-new/App/Controllers/PassesController.php(10): App\Models\Pass->getList() #1 [internal function]: App\Controllers\PassesController->indexAction(Array) #2 /var/www/wx-new/Lib/Controller.php(22): call_user_func_array(Array, Array) #3 /var/www/wx-new/Lib/Router.php(56): Lib\Controller->__call('index', Array) #4 /var/www/wx-new/Lib/Router.php(13): Lib\Router->handleRequest() #5 /var/www/wx-new/public/index.php(16): Lib\Router->__construct() #6 {main} thrown in /var/www/wx-new/App/Models/Pass.php on line 25_

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Sat 20 Mar 16:52:17 GMT 2021
Current date/time (ms):  1616259137
Repo git hash:           ef4d41cfbc2953043d5031fa4d8a9449280a1389
Raspberry Pi Model:      Raspberry Pi 2 Model B Rev 1.1
Architecture:            Linux raspberry-noaa 5.4.83-v7+ #1379 SMP Mon Dec 14 13                                                                                        :08:57 GMT 2020 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [22 Sat Mar 20 17:26:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/rec                                                                                        eive_noaa.sh "NOAA 19" NOAA-19-20210320-172609 /home/pi/raspberry-noaa-v2/tmp/or                                                                                        bit.tle 1616261169 944 66 Northbound E
  * [19 Sat Mar 20 18:05:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/rec                                                                                        eive_noaa.sh "NOAA 15" NOAA-15-20210320-180557 /home/pi/raspberry-noaa-v2/tmp/or                                                                                        bit.tle 1616263557 965 60 Northbound E
  * [21 Sat Mar 20 21:54:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/rec                                                                                        eive_noaa.sh "NOAA 18" NOAA-18-20210320-215438 /home/pi/raspberry-noaa-v2/tmp/or                                                                                        bit.tle 1616277278 948 33 Northbound W
  * [20 Sat Mar 20 20:13:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/rec                                                                                        eive_noaa.sh "NOAA 18" NOAA-18-20210320-201333 /home/pi/raspberry-noaa-v2/tmp/or                                                                                        bit.tle 1616271213 963 51 Northbound E
  * [23 Sat Mar 20 18:30:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/rec                                                                                        eive_meteor.sh "METEOR-M 2" METEOR-M-2-20210320-183021 /home/pi/raspberry-noaa-v                                                                                        2/tmp/orbit.tle 1616265021 947 69 Northbound E
---------------------------------------------
USB Device Map:
  Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fas                                                                                        t Ethernet Adapter
  Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 14.5G  0 disk
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 14.2G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       14657232 5386720   8642576  39% /
  devtmpfs          440244       0    440244   0% /dev
  tmpfs             473524       0    473524   0% /dev/shm
  tmpfs             473524    6296    467228   2% /run
  tmpfs               5120       4      5116   1% /run/lock
  tmpfs             473524       0    473524   0% /sys/fs/cgroup
  tmpfs             204800       0    204800   0% /var/ramfs
  /dev/mmcblk0p1    258095   46273    211822  18% /boot
  tmpfs              94704       0     94704   0% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:         947048 kB
  MemFree:           22112 kB
  MemAvailable:     830028 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
No supported devices found.

**Server information (please complete the following information):**
 - OS: Raspbian
 - SDR device Nooelec RTL-SDR SmarTee (Wasn't plugged in but not relevant)

jekhokie, 2021-03-22:
@pineapplee85 can you post the output here of the following commands:

`sqlite3 db/panel.db ".tables"`
`sqlite3 db/panel.db ".schema decoded_passes"`
`sqlite3 db/panel.db ".schema predict_passes"`

jekhokie, 2021-03-23:
@pineapplee85 I just updated the support script to output the values needed for the database schema - if you can, `git pull` and then run `./support.sh` and post the output again please?

pineapplee85, 2021-03-23:
Hi, I ran the install upgrade script again and it is working just fine. Twitter posting isnt but not a big deal.
Would you still want me to post the output

jekhokie, 2021-03-23:
@pineapplee85 depends on whether you are still having issues with accessing the webpanel (this error) - if not, don't worry about it, we can close this.

closed 2021-03-24 by jekhokie

lmarcuzzo, 2021-04-06:
Hello. I was having the same issue after upgrading. However, manually running db_migrations/update_database.sh seems to solve the problem.

Here is the output from support.sh after migration:

```
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer,
    foreign key(pass_start) references passes(pass_start));
```
And then after update_database.sh:

```
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));
```