jekhokie/raspberry-noaa-v2 #299: The parameter "FREQ_OFFSET" not given to rtl_

yvesclousier, 2021-03-21:
Very good job 👍 

**Describe the bug**
The parameter FREQ_OFFSET is missing in noaa_record and metor_record script.

exemple in noaa_record.sh 
 `timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -s 60k -g "${GAIN}" -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025`

the correct value is 
`timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -s 60k -g "${GAIN}" -p "${FREQ_OFFSET}" -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025`

jekhokie, 2021-03-22:
Very nice find, this is definitely a bug!

jekhokie, 2021-03-22:
Note to self: this is actually an issue for the NOAA receive script as well.

closed 2021-03-22 by jekhokie