jekhokie/raspberry-noaa-v2 #307: [Error]rtlsdr_m2_lrpt_rx.py does not take device as a parameter.

colinluthier, 2021-03-22:
``` # Arguments:
    #   1. Full path and name of stream file (including file extension)
    #   2. Gain to be used
    #   3. Frequency offset (PPM)
    #   4. Device Number
    stream_name = sys.argv[1]
    gain = float(sys.argv[2])
    freq_offset = int(sys.argv[3])
    device_number = int(sts.argv[4])'''

I think then we need to pass it:

if device = 0
self.connect((self.rtlsdr_source_0, 0), (self.rational_resampler_xxx_0, 0))
if device = 1
self.connect((self.rtlsdr_source_0, 0), (self.rational_resampler_xxx_1, 0))

jekhokie, 2021-03-24:
Do we know what needs to change here for certain? Reason I ask is because I'm hesitant to make the change since I myself cannot actually test this - can someone with the ability to use this script run a test using "device ID 1" and see if the proposed change works?

colinluthier, 2021-03-30:
I can as soon as I am done researching the E4000 bug. With two devices you can fm grab to data at the same time. Perhaps we have a better way in the long term but this is a short term solution.

dom-robinson, 2021-03-30:
Thing that gets me is that we *should* be able to grab the entire 137Mhz range all the time, and extract the sat passes from it by schedule, even with just one SDR card. 

I get why you are going for multi-dev switching. I also think all these things will get easier and easier as we get gnu radio in. I am working on NOAA APT capture-to-wav workflow, and the gnu radio block supports maybe a dozen SDR cards so it will really simplify compatibility - and gives a nice surface for debug if it needs it.

dom-robinson, 2021-09-29:
lets move this discussion to #437 since that is nearing PR.

closed 2021-09-29 by dom-robinson