jekhokie/raspberry-noaa-v2 #315: Display Photo Comments/Tweet/Discord

colinluthier, 2021-03-23:
wxtoimg provides tons of data in the comments of an image. If we could extract it we could send it as a Twitter/Discord comment or display it on the web viewer. Here is an example:

```Satellite: NOAA 18 Pass Start: 23 Mar 2021 18:19:09 GMT Pass Duration: 15:35 Elevation: 80 Azimuth: 287 Solar Elevation: 44.6 Direction: southbound Creation Time: 23 Mar 2021 18:23:54 GMT Satellite Type: NOAA Channel A: 1 (visible) Channel B: 4 (thermal infrared) Enhancement: HVCT Latitude: 38.296 Longitude: -122.307 Altitude: 0.0 Frequency: 137.9125 Audio File: /srv/audio/noaa/NOAA-18-20210323-111909.wav Map File: /home/pi/raspberry-noaa-v2/tmp/map/NOAA-18-20210323-111909-map.png Software: WXtoImg version 2.11.2 beta```

colinluthier, 2021-03-23:
sweet!
```pi@raspberrypi:/srv/images $ rdjpgcom  NOAA-18-20210317-221426-MCIR-precip.jpg
Satellite: NOAA 18
Pass Start: 18 Mar 2021 05:14:26 GMT
Pass Duration: 15:48
Elevation: 88
Azimuth: 82
Solar Elevation: -1.0
Direction: northbound
Creation Time: 18 Mar 2021 05:15:55 GMT
Satellite Type: NOAA
Channel A: 3/3B (mid infrared)
Channel B: 4 (thermal infrared)
Enhancement: MCIR-precip
Latitude: 38.296
Longitude: -122.307
Altitude: 0.0
Frequency: 137.9125
Audio File: /srv/audio/noaa/NOAA-18-20210317-221426.wav
Map File: /home/pi/raspberry-noaa-v2/tmp/map/NOAA-18-20210317-221426-map.png
Software: WXtoImg version 2.11.2 beta
```

colinluthier, 2021-03-23:
```sudo apt-get install libjpeg-progs```
 ```rdjpgcom  NOAA-18-20210317-221426-MCIR-precip.jpg```

dom-robinson, 2021-03-24:
top - love it. well done :)

jekhokie, 2021-03-24:
I'm curious about this - we already have a feature request in the queue to enable "customized push annotations", and most (if not all) of these attributes are already available to the annotation template in general. Is there something in particular you're looking to get out of this that we can't (or haven't already) exposed as part of the custom annotation template?

jekhokie, 2021-03-26:
Closing in favor of #252 as we already have all these attributes available to us (confirmed with Colin he’s ok with this approach).

closed 2021-03-26 by jekhokie