jekhokie/raspberry-noaa-v2 #331: [Error] Not getting any images

porkiedev, 2021-03-29:
Not receiving any images on my RPi 3b V2.0 with the latest release. I first tried on a micro-sd card that had rtl_tcp installed, I figured that was interfering, so I got a brand new card and installed only raspberry-noaa-v2 on it, still the same issue. I leave it overnight and get no images whatsoever.

(Not sure where the logs file is, otherwise I might even be able to solve this myself.)
**CURRENT SETTINGS.YML**

`---
# base station configurations
#   latitude: south values are negative
#   longitude: west values are negative
latitude: 37.695690
longitude: -97.473870

# time zone offset from UTC (for example, '-5' for US Eastern)
timezone_offset: -6

# ntp configurations
#   ntp_server - if you have a local server (e.g. stratum1), you can use this setting
#        to specify a hostname or ip address to communicate with the ntp server
ntp_server: ''

# test settings when running test scripts
#   test_gain - gain to use for scanner test scripts
#   test_sdr_device_id - device ID of the SDR device to be used for scanner test scripts
#   test_enable_bias_tee - whether to enable bias tee when running scanner test scripts
#   test_freq_offset - receiver frequency offset (PPM)
test_gain: 7.7
test_sdr_device_id: 0
test_enable_bias_tee: false
test_freq_offset: -6

# receiver settings
#   meteor_receiver - which receiver method to use (either 'rtl_fm' or 'gnuradio')
#     **WARNING**: 'gnuradio' does not work with certain SDR devices (e.g. it will not currently
#                  work with a RTL-SDR v3 dongle, as no image will be decoded from the bitstream)
meteor_receiver: 'rtl_fm'

# whether to schedule specific orbiting objects for capture
#   <satellite_name>_schedule - whether to schedule captures for the satellite
#   <satellite_name>_sdr_device_id - device ID of the SDR device to be used for recording for the satellite
#   <satellite_name>_freq_offset - receiver frequency offset (PPM) for the satellite capture
#   <satellite_name>_enable_bias_tee - whether to enable bias tee for the recording of the satellite
#   <satellite_name>_gain - gain setting for specific satellite captures
#   <satellite_name>_sun_min_elevation - threshold for sun elevation for specific satellite captures
#   <satellite_name>_sat_min_elevation - threshold for sat elevation for specific satellite captures
noaa_15_schedule: true
noaa_15_sdr_device_id: 0
noaa_15_freq_offset: -6
noaa_15_enable_bias_tee: false
noaa_15_gain: 7.7
noaa_15_sun_min_elevation: 6
noaa_15_sat_min_elevation: 30

noaa_18_schedule: true
noaa_18_sdr_device_id: 0
noaa_18_freq_offset: -6
noaa_18_enable_bias_tee: false
noaa_18_gain: 8.7
noaa_18_sun_min_elevation: 6
noaa_18_sat_min_elevation: 30

noaa_19_schedule: true
noaa_19_sdr_device_id: 0
noaa_19_freq_offset: -6
noaa_19_enable_bias_tee: false
noaa_19_gain: 7.7
noaa_19_sun_min_elevation: 6
noaa_19_sat_min_elevation: 30

meteor_m2_schedule: true
meteor_m2_sdr_device_id: 0
[... 259 lines cut ...]
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));`

**OS INFORMATION**
 - OS: Raspbian Buster (Latest version)
 - Python version (2.7.16)
 - SDR device (Nooelec NESDr SMArt v4)

As I said, I am currently running raspberry-noaa-v2 on a **brand new** raspbian buster os install. I made sure not to install anything except for git, but I am still receiving no images in the web panel or from what I can tell, the image output directory. (ansible/images_output?)

jekhokie, 2021-03-29:
@porkiedev check out the log file `/var/log/raspberry-noaa/output.log` and see if there are any clues in there?

Also, are you on the Discord chat server by chance? There are lots of people in there that can help troubleshoot a bit easier/real-time, as troubleshooting over the GitHub issue is not ideal.

porkiedev, 2021-03-30:
> @porkiedev check out the log file `/var/log/raspberry-noaa/output.log` and see if there are any clues in there?
>
> Also, are you on the Discord chat server by chance? There are lots of people in there [...]

Absolutely, discord is my life, and thanks for providing the directory for the log files. the only errors are from ./test_reception.sh (when I performed tests).

[
29-03-2021 00:59 ./scripts/schedule.sh INFO : TLE update not specified '-t' but no TLE files present - please re-run with '-t'
29-03-2021 01:03 ./test_reception.sh ERROR : Usage: ./test_reception.sh <frequency>. Example: ./test_reception.sh 90.3
29-03-2021 01:06 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : TLE update not specified '-t' but no TLE files present - please re-run with '-t'
29-03-2021 01:06 ./test_reception.sh ERROR : Usage: ./test_reception.sh <frequency>. Example: ./test_reception.sh 90.3
29-03-2021 01:06 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : TLE update not specified '-t' but no TLE files present - please re-run with '-t'
29-03-2021 02:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : TLE update not specified '-t' but no TLE files present - please re-run with '-t'
29-03-2021 13:17 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : TLE update not specified '-t' but no TLE files present - please re-run with '-t'
29-03-2021 13:43 ./test_reception.sh ERROR : Usage: ./test_reception.sh <frequency>. Example: ./test_reception.sh 90.3
29-03-2021 13:58 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : TLE update not specified '-t' but no TLE files present - please re-run with '-t'
]

I'll join the discord and if you're on, feel free to assist me directly, or I can find someone else that I'm sure will be willing to help.

``

closed 2021-03-30 by porkiedev