jekhokie/raspberry-noaa-v2 #345: Ability to Update Codebase Through Admin Interface of Webpanel

jekhokie, 2021-04-02:
It would be nice to enable the user to `git pull` (either latest, or tagged version - tagged version being default) to update their repository through the Admin interface of the webpanel, allowing the user to avoid all of the command-line nonsense. This would obviously only be applicable for upgrades (not necessarily first-time installs of course).

dom-robinson, 2021-04-07:
YESS!!! :) A few options like ‘update to latest master’ ‘set to tag release’ ‘auto update to master’ etc? Would be awesome.

If there was a way to also do something for us tinkerers to Re sync with Master while also maintaining any local changes we are working on that would be cool, but i can see that being a bit case-specific perhaps?

dom-robinson, 2021-05-12:
[image]

Rough version working in dev. Some way to go before i push anything, but can now edit the settings.yml in a new 'settings' tab on my browser which gives me a simple text editor with settings.yml loaded, and a way to post it back to the machine.

- [ ] move my work from hacked in /svr/www.... back into the source /raspberry-noaa-v2/webpanel and add the src files for the editor capability into the right place.

- [ ] script in the copying for the resources and setting permissions on the writable settings.yml 

i am thinking of having one  copy of the file that gets written and copied back to the pi, and then a second stage where user 'executes update' and this will copy the newly pushed / edited settings into the proper /config/ folder and launches the ./install_and_upgrade.sh (and provides progress or - as @colinluthier suggested - streams the progress of the update script back to the browser - handy for progress AND handy for debug). This could even allow for settings.yml to be garbage-rotated so the user could 'undo' or 'rollback' 3 or 4 updates to settings.yml if they wanted to.

a few days until i can get to this again, but worth sharing on this issue i think...

dom-robinson, 2021-09-29:
Argh the security holes. I began to hate my modification since i knew as soon as i put it out in the wild it would rile all the info-sec folks into a #flamewar at me :) So i backed away and deleted it.

I may return to this since it does have an 'obviousness' to it, but on reflection the 'right' way to do this is actually to move teh settings.yml settings into the DB and to create a web-admin tool that changes the DB settings and on save writes out a new settings.yml, and auto fires the install scripts, providing feedback to the UI that the scripts are progressing and have either run correctly or reports what has failed.... and all that for the sake of not using ssh... 

For now i am not jumping with excitement but would welcome a good DB / UX programmer picking it up since i know a lot of folks find ssh that bit more difficult, and in particular those who use the images that are being generated might welcome a UI in the webpanel for admin purposes.

Flagging 'help wanted..'