jekhokie/raspberry-noaa-v2 #346: Airspy support and Airspy + RTL-SDR combo

Akito01, 2021-04-02:
It would be nice if Airspy will be supported in future updates. As I'm reading the scripts now, it only relies on and supports RTL-SDR.

As an extra option it would be nice to support dual RTL-SDR or a combo of Airspy and RTL-FM.
- An option could be to always first use the Airspy and if occupied, then use RTL-FM.

- It would be nice as well for the above to add a checkbox to whether allow or disallow the use of both devices when Meteor and NOAA19 overlap too much (and perhaps be able to set how much overlap should occur before this is activated.) or to only use both devices for NOAA only.

- An other option could be to always use Airspy for Meteor and RTL-FM for NOAA or vice versa.

colinluthier, 2021-04-05:
I am working on the second part of that. I have a script that provides a noise rating to an image. Once we can rate the image we can set a display threshold. Right now I need to do the noise rating on a png. I just pushed a fix to make sure the pristine is outputting a png.

colinluthier, 2021-04-05:
Dom, is working on multiple radio support and we might get that pushed this week.

dom-robinson, 2021-04-06:
New incoming workflow introduces GNURADIO based nova wav capture and should support all these hardware devices:

dom-robinson, 2021-04-07:
Hmm - images attached to email replies here don’t get through: here is that list >>
[image]

b3rtdb, 2021-04-21:
Any progress on this? Can't wait to use my Airspy instead of the rtl-sdr v3 for the NOAA v2 setup :-)

dom-robinson, 2021-05-29:
> Any progress on this? Can't wait to use my Airspy instead of the rtl-sdr v3 for the NOAA v2 setup :-)

Just clearing a backlog of PRs and one of my draft PRs is the GNURadio for NOAA - that is working on my dev, but needs testing on a clean build: Hoping to get to that in the next few days now!

dom-robinson, 2021-09-29:
The PR for gnuradio is nearing a merge - https://github.com/jekhokie/raspberry-noaa-v2/blob/aug21-merging-gnuradio/scripts/audio_processors/README.md 

This will open the door for users to start to develop thier own hardware interfaces for non-rtl-sdr cards.

Im going to close this thread for now and ask folks to comment / beta and input on the #437 PR please...

closed 2021-09-29 by dom-robinson