jekhokie/raspberry-noaa-v2 #347: Sync NOAA map overlay to create animated gif from passes

Crosswalkersam, 2021-04-02:
Until now, I used Gimp to overlay ca. 10 images manually and then created a gif showing the speed and direction of clouds.

However, this is very time consuming and far from perfect. Would it be possible to have the script line up some images using the map overlay and then create a gif of the last, for example 10 images?

dom-robinson, 2021-04-07:
When i added the ‘animation’ video this was exactly what i was hoping to achieve, but i played with wxproj (the part of wxtoimg that supposedly lets you align the images) and for the life of me the ‘-b’ setting (which is supposed to set the corners of the projection around the GS location) just would not do anything logical or consistent in the same way the GUI side of wxtoimg did.

If you have been doing this manually with gif alignment in GIMP i would be interested to know if you think it is scriptable since if so i totally agree that this should be a feature we should add - very happy to help!

dom-robinson, 2021-09-29:
Woudl really love it if someone can find a way to make wxtoimg ALIGN animation frames - everything else in the workflow is there, but the Command Line WXTOIMG '-b' setting is unfathomable...

colinluthier, 2022-01-28:
@dom-robinson Is it not as easy as following this example?

```
To build up an animated (AVI) movie from a series of GOES recordings:
wxtoimg goes9-06.wav goes9.avi
wxtoimg goes9-09.wav goes9.avi
wxtoimg goes9-12.wav goes9.avi
wxtoimg goes9-15.wav goes9.avi
wxtoimg goes9-18.wav goes9.avi
```

tokoivun, 2022-04-09:
> @dom-robinson Is it not as easy as following this example?
> 
> ```
> To build up an animated (AVI) movie from a series of GOES recordings:
> wxtoimg goes9-06.wav goes9.avi
> wxtoimg goes9-09.wav goes9.avi
> wxtoimg goes9-12.wav goes9.avi
> wxtoimg goes9-15.wav goes9.avi
> wxtoimg goes9-18.wav goes9.avi
> ```

GOES is geostationary, NOAA's are on polar orbit. Field of view depends on the pass, hence the need for alignment.

colinluthier, 2022-04-13:
Thank you @tokoivun I hadn't even noticed that example was GOES.