jekhokie/raspberry-noaa-v2 #357: METEOR and TMPFS depending on free memory

Clnmt, 2021-04-07:
TMPFS is used for METEOR pass only if the total memory is 2GO or more, I think we should check the free memory instead. 

🎯 **Goal**
- This change would allow more users to execute METEOR operations on memory, for better performance and to limit SD card use.

💡 **Possible solutions**
 - Use 1000M as TMPFS folder size in any case ; and
 - On execution, if free memory is > 600M then use the TMPFS folder
  
📋  **Steps to solve the problem**
*   I can submit a PR if you agree.

dom-robinson, 2021-04-07:
I recall noticing that the test for this was not steering the processing to RAM and thinking at the time it would be good to look deeper - I totally agree that any processing that can be done in RAM should be - those SD cards do have a limited shelf-life! 

It’s really up to Justin to confirm the PR but I am fully in support: thankyou!

dom-robinson, 2021-09-29:
looks like it got worked in - closing for now :)

closed 2021-09-29 by dom-robinson