jekhokie/raspberry-noaa-v2 #366: Do you for planning add support to process the ISS(ZARYA) passes? Thx

lyovav, 2021-04-15:
ISS(ZARYA)

dom-robinson, 2021-04-15:
Not sure if Justin plans to bring this back in, but very early versions did support it, however in general the results were erratic and produced a lot of 'noise' - there is, however, no reason not to add it as an option for those who want it i guess...

lyovav, 2021-06-22:
[image]

surfrod, 2021-06-22:
Would love to see it added back in, seems the ISS ham radios are more stable after the last EVA to repair everything.  These SSTV events seem to happen a few times a year.

surfrod, 2021-06-22:
I started looking into adding it, but having never looked at the v2 code base before it'll be quite a learning process

lyovav, 2021-06-22:
> Would love to see it added back in, seems the ISS ham radios are more stable after the last EVA to repair everything. These SSTV events seem to happen a few times a year.

So events are not regular but unique, unfortunately you have to install the 1st at the time of the event as there is no more automatic script for this event. In my area, 1-2 images are accepted in 1 pass, so sometimes I have to re-process files with the RX-SSTV program.

dom-robinson, 2021-09-29:
happy to look at PRs of workflows to bring it back in. At the moment i personally think its a little 'just cos we can' rather than being weather satellite specific and i worry that if we do go down this path we might turn RN2 into a general SDR scheduled capture recevier and decoder, and that seems a bit ambitiously large. Equally i realise that SSTV is an APT signal and there is some logic to including it..

Be great if someone can generate some working examples we can explore together...

colinluthier, 2022-01-28:
Writing a dedicated standalone app using this demodulator might be an easier option:
https://github.com/reynico/pd120_decoder/tree/master/pd120_decoder

The workflow would be schedule the job
The job would fire up rtl-fm fo the capture
pd120_decoder would generate the image

You would have to create you own web page and database to hold the images....

closed 2024-12-02 by MihajloPi