jekhokie/raspberry-noaa-v2 #370: [Error] Autogain option not working

Crosswalkersam, 2021-04-18:
Hello,

after feature request #165 the automatic gain option was added. I tried setting the gain in config/settings.yml to 0 to enable it. The webinterface detects it and shows "gain: automatic", however the receiver scripts /scripts/audio_processors/noaa_record.sh and /scripts/audio_processors/meteor_record_rtl_fm.sh still seem to interpret it as a gain of 0dBm, resulting in no useful image output if the gain is set to 0. 

When forcing automatic gain by removing the -g attribute from the scripts named above, autogain works and I can get images.

cdtush, 2021-04-24:
So I  am a noob at programming but couldnt something like this wor
```
if [ ${GAIN} == 0 ]; then
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -p "${FREQ_OFFSET}" -s 60k  -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025
else
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -p "${FREQ_OFFSET}" -s 60k -g "${GAIN}" -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025
fi
```

cdtush, 2021-04-25:
I know my code is garbage.  I am not a programmer. But I tested this and it works.   
I did the following in noaa_record.sh

```
if [ ${GAIN} == 0 ]; then
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -p "${FREQ_OFFSET}" -s 60k  -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025
else
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -p "${FREQ_OFFSET}" -s 60k -g "${GAIN}" -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025
fi
```
and  the following in meteor_record_rtl_fm.sh

```
if [ ${GAIN} == 0]; then
log "Recording at ${METEOR_FREQ} MHz..." "INFO"
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -M raw -f "${METEOR_FREQ}"M -p "${FREQ_OFFSET}" -s 288k | $SOX -t raw -r 288k -c 2 -b 16 -e s - -t wav "${OUT_FILE}" rate 96k >> $NOAA_LOG 2>&1
else
log "Recording at ${METEOR_FREQ} MHz..." "INFO"
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -M raw -f "${METEOR_FREQ}"M -p "${FREQ_OFFSET}" -s 288k -g "${GAIN}" | $SOX -t raw -r 288k -c 2 -b 16 -e s - -t wav "${OUT_FILE}" rate 96k >> $NOAA_LOG 2>&1
fi
```

dom-robinson, 2021-05-12:
Nice - yes this needs sorting. IMHO we should scrap autogain and declare it 'not-supported' in RN2 and default to 40.

Nice code solution from @cdtush if we do continue it.

cdtush, 2021-05-14:
Auto gain has a place.  And having it as an option does not hurt anything, other than taking up a few Bytes of storage.  If I was better at coding,  (I may look into it as I have free time) would be this:  
```
if [ ${GAIN} == auto ]; then
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -p "${FREQ_OFFSET}" -s 60k  -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025
else
timeout "${CAPTURE_TIME}" $RTL_FM -d ${SDR_DEVICE_ID} ${BIAS_TEE} -f "${freq}"M -p "${FREQ_OFFSET}" -s 60k -g "${GAIN}" -E wav -E deemp -F 9 - | $SOX -t raw -e signed -c 1 -b 16 -r 60000 - "${OUT_FILE}" rate 11025
fi
```

But I have no Idea if I can use that variable in here,  Using the 0 works,  but it is far from the most effective way to do it.  

Personally I run my SDR now with the autogain option only.  I have found that I tend to get a better image through different weather conditions.

dom-robinson, 2021-09-29:
Interesting - Tagging @colinluthier since i think he had a play with autogain a while ago and if this represents a 'fix' we should probably queue this up to include...

colinluthier, 2022-01-28:
Except, I am really slow... This was added at some point so I am closing. There is some interesting legacy code in receive_noaa.sh

```
gain=$GAIN
        if [ $(echo "$GAIN==0"|bc) -eq 1 ]; then
          gain='Automatic'
        fi
```

I think it is now unused. I will note it here for future reference.

closed 2022-01-28 by colinluthier