jekhokie/raspberry-noaa-v2 #371: Specified SDR device in config

pineapplee85, 2021-04-20:
Hi there,
Was just wondering if it would be worth adding a "Device = " string in the config file to allow the user to specify a specific SDR receiver? eg. _Device= 00001001_

The RTL-SDR USB stick may be reprogrammed using the following command: _rtl_eeprom_ -s _00001001_ 
(Please note: it is advised to provide a serial number that begins with 00001 eg: 00001002  00001003 ect and that re-programming any read only memory comes with its risks, which may render your device unusable.)

This would allow the user to have multiple SDRs connected to their Pi and different antennas enabling the use of software such as SpyServer or others, ensuring they do not interfere / raspberry-noaa does not interfere with the user's software.

It could also be used to allow different antenna and LNA styles to be used based on the satellite type. (eg. V-Diple for NOAA, QFH + LNA for Meteor)

I believe the rtl_fm command can have a device specified with the flag:  -d [serial number]  eg:  _rtl_fm -d 00001001_

Thought this might be worth mentioning as I just found out how to do it
Ethan :)

dom-robinson, 2021-04-21:
Hey Ethan - i think what you need is in part already in there?  

eg [./config/settings.yml.sample](https://github.com/jekhokie/raspberry-noaa-v2/blob/master/config/settings.yml.sample) line 41 `noaa_15_sdr_device_id: 0`

However i get the advancement of your idea so multiple devices could be a pool for arbitrary use on a pass by pass basis rather than on a sat by sat one where devices are (as they currently are) tied to each satellite.

pineapplee85, 2021-04-21:
Hi,
Sorry, my bad I didn't see that. Does this utilise the device serial? eg. 00001002 would be acceptable? 
Ethan

pineapplee85, 2021-04-24:
Closed due to testing working out.

closed 2021-04-24 by pineapplee85