jekhokie/raspberry-noaa-v2 #374: [Error] "flip_meteor_image:" does not work

Krolykke, 2021-04-21:
**Describe the bug**
The setting "flip_meteor_image:" does not work, nothing happens no matter if its set to true or false

**Screenshots and/or logs**
n/a

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: ons 21 apr 06:33:10 CEST 2021
Current date/time (ms):  1618979590
Repo git hash:           94817fe43d2a7594d0da9305ee0601c8b15cc7ff
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.4
Architecture:            Linux RaspiSat 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [172	Thu Apr 22 09:44:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210422-094407 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619077447 921 58 Southbound W
  * [170	Thu Apr 22 19:40:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210422-194042 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619113242 965 38 Northbound W
  * [150	Wed Apr 21 22:35:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210421-223527 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619037327 948 43 Northbound W
  * [149	Wed Apr 21 20:54:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210421-205450 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619031290 958 42 Northbound E
  * [171	Thu Apr 22 08:03:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210422-080354 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619071434 931 29 Southbound E
  * [161	Thu Apr 22 17:54:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210422-175458 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619106898 877 24 Northbound E
  * [157	Wed Apr 21 18:14:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210421-181436 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619021676 885 27 Northbound E
  * [165	Thu Apr 22 20:43:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210422-204309 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619116989 902 35 Northbound E
  * [162	Thu Apr 22 19:33:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210422-193330 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619112810 942 80 Northbound W
  * [166	Thu Apr 22 22:23:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210422-222324 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619123004 955 52 Northbound W
  * [158	Wed Apr 21 19:53:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210421-195346 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619027626 972 69 Northbound W
  * [173	Thu Apr 22 17:55:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210422-175512 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619106912 826 21 Northbound E
  * [169	Thu Apr 22 17:59:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210422-175951 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619107191 978 48 Northbound E
  * [167	Thu Apr 22 08:06:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210422-080622 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619071582 973 63 Southbound E
  * [159	Thu Apr 22 08:06:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210422-080607 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619071567 883 33 Southbound E
  * [174	Thu Apr 22 19:33:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210422-193349 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619112829 976 87 Northbound E
  * [156	Wed Apr 21 10:04:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210421-100406 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1618992246 901 43 Southbound W
  * [155	Wed Apr 21 08:23:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210421-082346 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1618986226 949 40 Southbound E
  * [148	Wed Apr 21 12:42:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210421-124210 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619001730 895 32 Southbound W
  * [160	Thu Apr 22 09:45:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210422-094559 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619077559 951 50 Southbound W
  * [147	Wed Apr 21 11:01:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210421-110116 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1618995676 952 58 Southbound E
  * [163	Thu Apr 22 10:49:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210422-104922 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619081362 950 48 Southbound E
  * [168	Thu Apr 22 09:47:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210422-094716 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619077636 906 30 Southbound W
  * [164	Thu Apr 22 12:30:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210422-123014 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1619087414 915 38 Southbound W
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: -2
  noaa_15_enable_bias_tee: true
  noaa_15_gain: 25.4
  noaa_15_sun_min_elevation: -90
  noaa_15_sat_min_elevation: 20
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: -2
  noaa_18_enable_bias_tee: true
  noaa_18_gain: 25.4
  noaa_18_sun_min_elevation: -90
  noaa_18_sat_min_elevation: 20
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: -2
  noaa_19_enable_bias_tee: true
  noaa_19_gain: 25.4
  noaa_19_sun_min_elevation: -90
  noaa_19_sat_min_elevation: 20
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: -2
  meteor_m2_enable_bias_tee: true
  meteor_m2_gain: 25.4
[... 54 lines cut ...]
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
[... cut ...]  integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

**Server information (please complete the following information):**
 - OS: Raspbian
 - Python version: Python 2.7.16 (default, Oct 10 2019, 22:02:15) [GCC 8.3.0] on linux2]
 - SDR device RTL-SDR v3

**Additional context**
n/a

dom-robinson, 2021-05-12:
Noted - bubbling up the to-do list :)

dom-robinson, 2021-09-29:
I think there is an update that made it into one PR then not into another. There is a new PR coming out and I am pretty confident that the flip fix made it through - once that goes to master (hopefully fairly soon now) i think this issue will get fixed once and for all... Closing for now, but we can reopen if the fix doesn't materialist..

closed 2021-09-29 by dom-robinson