jekhokie/raspberry-noaa-v2 #377: [Error] Failed Decode

pineapplee85, 2021-04-29:
NOAA output images are not recording / decoding correctly. When listening back to the audio file it sounds like the metronome is fast. 
Meteor is fine.

[Raspiv2 Logs.zip](https://github.com/jekhokie/raspberry-noaa-v2/files/6400827/Raspiv2.Logs.zip)

**Support Output**

> =============================================
Details about environment
=============================================
Current local date/time: Thu 29 Apr 19:27:51 BST 2021
Current date/time (ms):  1619720871
Repo git hash:           94817fe43d2a7594d0da9305ee0601c8b15cc7ff
Raspberry Pi Model:      Raspberry Pi 2 Model B Rev 1.1
Architecture:            Linux garage-noaa 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08                                                                                        :57 GMT 2020 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [24 Thu Apr 29 21:41:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/rec                                                                                        eive_noaa.sh "NOAA 18" NOAA-18-20210429-214108 /home/pi/raspberry-noaa-v2/tmp/or                                                                                        bit.tle                                                               1619728868                                                                                         944 73 Northbound E
  * [34 Thu Apr 29 19:35:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/rec                                                                                        eive_meteor.sh "METEOR-M 2" METEOR-M-2-20210429-193514 /home/pi/raspberry-noaa-v                                                                                        2/tmp/orbit.tle                                                               16                                                                                        19721314 939 80 Northbound E
  * [29 Thu Apr 29 21:41:00 2021 a pi] ->
  * [19 Thu Apr 29 19:35:00 2021 a pi] -> /home/pi/raspberry-noaa/receive_meteor                                                                                        .sh "METEOR-M 2" 137.1000 METEOR-M220210429-193514 /home/pi/raspberry-noaa/predi                                                                                        ct/weather.tle 1619721314 939 80
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: -6
  noaa_15_enable_bias_tee: true
  noaa_15_gain: 17
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: -6
  noaa_18_enable_bias_tee: true
  noaa_18_gain: 17
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: -6
  noaa_19_enable_bias_tee: true
  noaa_19_gain: 17
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: -6
  meteor_m2_enable_bias_tee: true
  meteor_m2_gain: 23
  meteor_m2_sun_min_elevation: 6
  meteor_m2_sat_min_elevation: 30
---------------------------------------------
USB Device Map:
  Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fas                                                                                        t Ethernet Adapter
  Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 14.5G  0 disk
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 14.2G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       14657232 5505420   8523876  40% /
  devtmpfs          440244       0    440244   0% /dev
  tmpfs             473524       0    473524   0% /dev/shm
  tmpfs             473524   12252    461272   3% /run
  tmpfs               5120       4      5116   1% /run/lock
  tmpfs             473524       0    473524   0% /sys/fs/cgroup
  tmpfs             204800       0    204800   0% /var/ramfs
  /dev/mmcblk0p1    258095   46273    211823  18% /boot
  tmpfs              94704       0     94704   0% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:         947048 kB
  MemFree:          208524 kB
  MemAvailable:     752168 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00001002
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max in                                                                                        t, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer,                                                                                         has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, h                                                                                        as_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_his                                                                                        togram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

**Server information (please complete the following information):**
 - OS: Raspbian
 - SDR device Nooelec NesdrSmarTee

pineapplee85, 2021-05-01:
Update:
Now not getting passes scheduled (01/05/21)

dom-robinson, 2021-05-12:
still not working?

Might be worth checking the SDR is performing as expected - if you haven't do have a check on the "Reception" part of the docs here  [https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/troubleshooting.md#reception](https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/troubleshooting.md#reception) 

Ultimately if that can capture the audio ok, then the scripts should call the same processes to capture. 

Always worth giving the ./install_and_upgrade.sh a run every now and then, although i cant imagine you are not doing so after any adjustment to the settings.yml... :)

dom-robinson, 2021-09-29:
Happy to reopen, but I am cleaning up the issues lists a little and I am working on assumption you either got it working or abandoned?

closed 2021-09-29 by dom-robinson