jekhokie/raspberry-noaa-v2 #378: Docker container - to preserve the OS

tubalainen, 2021-05-06:
**Please Describe the problem to be solved**
To ease management, installation and upgrades as well as keeping the original OS intact, please consider creating a docker container for this fantastic hack! :)

Thanks for your hard and dedicated work! <3

dom-robinson, 2021-05-12:
Good idea. I also think a ready to configure image might be a good thing to produce at some point. 

There is a bit of  a backlog of PRs and issues we are going to filter and prioritise over the next few weeks. Personally i think we need to debug a few things and add a few more features that have been waiting to come online for a while before we cut images / dockercontainers - else we will be adding more fire to the fire lol!

But i agree its a good idea - can you remind / bug us in a month or two?

tubalainen, 2021-06-21:
Gentle poke :)

I have been working with docker for a few years and its fantastic. I will try to take a initial stab at getting it going.
Anyone wanna collaborate? :)

//T

dom-robinson, 2021-09-29:
Very happy to throw some input in - i did have a crack, but tbh it was distracting me so much from progressing the 'urgent' to do lists for RN2 i parked it and then realised i wiped that SD card (doh!)

I had more or less got it working but the external dockerfile was getting more complex to manage on each install and i worked out that going through the docker setup was taking longer than a clean SD card install, so for my own purposes of dev it was becoming counter-efficient. 

That said happy to throw thoughts in here if you progress and need some input!

rhaamo, 2021-11-03:
A docker image would be really useful, right now I already have a dedicated server for radio stuff and I don't want to spend money on yet-another-pi and it's not possible to install at all without having to spend hours digging into the install part.

sachasmart, 2024-03-20:
Watching this as well! Happy to give it a go if this has not progressed?