jekhokie/raspberry-noaa-v2 #384: Getting very noisy images

mk1blacklimited, 2021-05-12:
**Describe the bug**
Probably not a bug. :)

**Screenshots and/or logs**
All my images are very noisy, this is the best I've captured so far.
[image]

Spectrogram:
[image]

settings:

---
# base station configurations
#   latitude: south values are negative
#   longitude: west values are negative
latitude: 59.945454
longitude: 11.009236

# time zone offset from UTC (for example, '-5' for US Eastern)
timezone_offset: 1

# ntp configurations
#   ntp_server - if you have a local server (e.g. stratum1), you can use this setting
#        to specify a hostname or ip address to communicate with the ntp server
ntp_server: ''

# test settings when running test scripts
#   test_gain - gain to use for scanner test scripts
#   test_sdr_device_id - device ID of the SDR device to be used for scanner test scripts
#   test_enable_bias_tee - whether to enable bias tee when running scanner test scripts
#   test_freq_offset - receiver frequency offset (PPM)
test_gain: 7.7
test_sdr_device_id: 0
test_enable_bias_tee: true
test_freq_offset: -6

# receiver settings
#   meteor_receiver - which receiver method to use (either 'rtl_fm' or 'gnuradio')
#     **WARNING**: 'gnuradio' does not work with certain SDR devices (e.g. it will not currently
#                  work with a RTL-SDR v3 dongle, as no image will be decoded from the bitstream)
meteor_receiver: 'rtl_fm'

# whether to schedule specific orbiting objects for capture
#   <satellite_name>_schedule - whether to schedule captures for the satellite
#   <satellite_name>_sdr_device_id - device ID of the SDR device to be used for recording for the satellite
#   <satellite_name>_freq_offset - receiver frequency offset (PPM) for the satellite capture
#   <satellite_name>_enable_bias_tee - whether to enable bias tee for the recording of the satellite
#   <satellite_name>_gain - gain setting for specific satellite captures
#   <satellite_name>_sun_min_elevation - threshold for sun elevation for specific satellite captures
#   <satellite_name>_sat_min_elevation - threshold for sat elevation for specific satellite captures
noaa_15_schedule: true
noaa_15_sdr_device_id: 0
noaa_15_freq_offset: -6
noaa_15_enable_bias_tee: true
noaa_15_gain: 0
#noaa_15_gain: 7.7
noaa_15_sun_min_elevation: 6
noaa_15_sat_min_elevation: 30

noaa_18_schedule: true
noaa_18_sdr_device_id: 0
noaa_18_freq_offset: -6
noaa_18_enable_bias_tee: true
noaa_18_gain: 0
#noaa_18_gain: 8.7
noaa_18_sun_min_elevation: 6
noaa_18_sat_min_elevation: 30

noaa_19_schedule: true
[... 1515 lines cut ...]
  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

**Server information (please complete the following information):**
 - OS: Raspbian
 - Python version 2.7.16
 - SDR device RTL-SDR v3

**Additional context**
Just started this, and I've reinstalled the SD-card twice. Seems very much like this https://github.com/jekhokie/raspberry-noaa-v2/issues/96 but a reinstall didn't help. I've tried adjusting the gain from 2/3 to max, now it's at auto. I do get lots of files in /srv/*, e.g. 
ls -l /srv/audio/noaa/
total 88868
-rw-r--r-- 1 pi pi  2628262 May 11 18:18 NOAA-15-20210511-181637.wav
-rw-r--r-- 1 pi pi  2628262 May 11 19:57 NOAA-15-20210511-195550.wav
-rw-r--r-- 1 pi pi 21195938 May 11 12:27 NOAA-18-20210511-120415.wav
-rw-r--r-- 1 pi pi 22011800 May 11 22:15 NOAA-18-20210511-215956.wav
-rw-r--r-- 1 pi pi 20711238 May 11 17:54 NOAA-19-20210511-173928.wav
-rw-r--r-- 1 pi pi 21814562 May 11 19:35 NOAA-19-20210511-191938.wav

I've only got a rabbits ear antenna, so that might explain it all? I've tried tuning it to FM reception just to make sure it actually is ok, and FM radio reception works.

dom-robinson, 2021-05-12:
I can see you are just about getting a signal (the vertical dark stripes on the spectrogram are the clue). But its very weak. To compare i have pushed an 'average' quality one from a pass here this morning:

[image]

So its a simple solution i think: The gains are defaulted to 'auto' which essentially doesnt really work. 

in settings.yml : ```noaa_15_gain: 0``` etc

Set the gains in each satellite's settings up to around 38 or 42 (they are all around 7/8 at the moment) 

Note @colinluthier : I think we should probably lift the default gains right up to around 40 in a future update to the repo since quite a few folks come round with this issue.

mk1blacklimited, 2021-05-12:
Thanks for replying! :)

I think I mentioned (among all the other text...:D) I've tried adjusting the gain, both at 2/3 and to max. I did however try what you recommended(-ish), and set the gain to 40.2. This is what I got now, and while it looks a little better in the middle of the picture, it's still not quite what I'd like to see:

[image]

This is the spectrogram:

[image]

cdtush, 2021-05-12:
What type of antenna are you using?

mk1blacklimited, 2021-05-12:
The rabbits ear antenna that came with the RTL-SDR dongle.

cdtush, 2021-05-13:
Each leg of the rabbit ear needs to be 21.02 inches (53.4cm).  In a v at 120 degrees.   The center of the V should point North.
The V be laid flat (Both legs should be parallel to ground.
I found 44.3 gain to work about the best for me.  But I ended up altering some code and I use Auto gain with a turnstile antenna.

mk1blacklimited, 2021-05-13:
No worries, work trumps hobby. :)

The antenna is configured as you describe, except the v is pointing south - I've read it doesn't matter if it's pointing north or south?

I've moved the antenna to the top of my roof since my last posting, and increased the gain to 40.2, and the images have improved:

[image]

I've just changed the gain to 44.5 (closest I got to what rtl_test presents as available), so exited to see if that helps even more. :)

I actually got an image from METEOR as well, that's a first:

[image]

mk1blacklimited, 2021-05-14:
Max gain now, getting there:

[image]

[image]

I guess I could turn the antenna so the v is facing north, but not sure if it would make reception better?

A QFH antenna would probably be the next thing? I really shouldn't be using so much time on this (but hey, it's fun!), so I really could use some advice about buying an antenna instead of making one - recommendations?

cdtush, 2021-05-14:
I went looking for a prebuilt antenna, about 2 months ago,  Never really found one that I could not make for WAY less than what they would cost.  Ended up with a turn-style antenna,  Mainly for the ease of building.  I used 1/2 inch (12mm) copper pipe, and RG6 tv antenna (satellite/cable)  wiring.  With this setup I get pretty consistent images
[image]

[image]

mk1blacklimited, 2021-05-21:
That's interesting, the QFH-antenna build seems a bit daunting to me. :)

With my V-style antenna I'm now quite happy with these images (one of the better ones):

[image]

I have maxed out the gain, but should I try to lower it some to further improve the image, or is that no point?

Did you build your antenna from a guide? Link if you have one :)

cdtush, 2021-05-21:
I Used this image, as my guide.   Elements and phasing cable are 534mm long.  I did not use a matching cable.
[image]

Kamal-4S7AB, 2021-07-07:
A properly made QFH will be a very good solution for noise free picture. I use one and works great even decoding Meteor images

dom-robinson, 2021-09-29:
Looks like 'team RN2' got you sorted! Closing now, but good to see the success!

closed 2021-09-29 by dom-robinson