jekhokie/raspberry-noaa-v2 #390: Pillow install and new RaspbiOS

dinoj-bg, 2021-05-27:
If You have some problems  with fresh install  on latest RaspiOS Please note that You can not compile Pillow.
You need some dependecies to install first. Use https://pillow.readthedocs.io/en/latest/installation.html or just do
sudo apt install python3-pil and python3-pil.imagetk libtiff5-dev libjpeg-dev libopenjp2-7-dev zlib1g-dev     libfreetype6-dev liblcms2-dev libwebp-dev tcl8.6-dev tk8.6-dev python3-tk     libharfbuzz-dev libfribidi-dev libxcb1-dev
I also do not have libjpeg9, so You need manually edit ~/raspberry-noaa-v2/ansible/roles/common/tasks/dependencies.yml to remove libjpeg9 and correct  libjpeg9-dev to libjpeg-dev 
It latest version PHP is now 7.3 version so You also need to edit 3 files 
~/raspberry-noaa-v2/ansible/roles/webserver/templates/nginx_default.j2
~/raspberry-noaa-v2/ansible/roles/webserver/templates/nginx_tls_default.j2
~/raspberry-noaa-v2/ansible/roles/webserver/tasks/main.yml
and change php7.2 to php7.3
and final word if You do not have predict You need to download it from debian ftp repo http://ftp.debian.org/debian/pool/main/p/predict/  and install it with dpkg -i and do apt -f install for dependencies.  Before download chech Your architecture /armhf for 32-bit or arm64 for 64 bit/ 
I hope this will help.

dom-robinson, 2021-05-29:
@colinluthier @Cadair @jekhokie - Makes me wonder if we need to declare a particular OS release as a 'reference install' base. At the moment i think we 'hope' that everyone is on a slightly older Raspbian. 

There is a [thread](https://github.com/jekhokie/raspberry-noaa-v2/issues/378) about containers too - that might help, although without some checking it won't be a conclusive panacea...

colinluthier, 2021-05-31:
We should make a note in the readme that says to download the last official release and not the most recent build.