jekhokie/raspberry-noaa-v2 #393: Notification LED via GPIO

pagan27, 2021-05-30:
Hi!
Can you add this feature to the program?
Notification LED via GPIO.
Example: two minutes before when the satellite passes, the LED flashes for one second every ten seconds . When the recording, decoding is started, the LED is blinking until the process end. When everything is finished, the output files are created, the LED is lighting two minutes.
Thanks for your response.

dom-robinson, 2021-06-07:
nice idea, but i think each GPIO setup would end up pretty bespoke to each user no?

Quite a nice thread [here](https://thepihut.com/blogs/raspberry-pi-tutorials/27968772-turning-on-an-led-with-your-raspberry-pis-gpio-pins), and if i was going to do it i would simply the requirement a little by calling the GPIO light from the scheduled 'at' tasks so as they run the LED is flashed.  I think the 'two minutes before' might add some complication in that you would then have to have another at task schedueld or the scehduled capture sequence effectively would have to start 2 min earlier and that might cut off more overlapping captures...

Cadair, 2021-06-07:
we could allow some pre and post pass scripts to be called, then custom ones for gpio could be written?

paradox-productions, 2021-06-08:
Custom scripts would be really helpful. Currently testing on a pi-top case with a small screen on it, this would be useful for the screen on this also

stevecrow74, 2021-06-18:
I once had this idea running, an LED would light at begining of pass than end at the end off pass, i had the script  looking out for when the rtl was running, and if it was it would switch on a gpio. here's the script below.

#!/usr/bin/python
import RPi.GPIO as GPIO
import os
import subprocess
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(18, GPIO.OUT)#Sets the GPIO pin to be an output pin
GPIO.output(18, False) #Makes sure all LED's are at the same state (off)

RTL_SDR = b"/usr/local/bin/rtl_fm"

output = subprocess.check_output(["ps", "-ef"])

if RTL_SDR in output:
     GPIO.output(18, True) #Turn on LED
else:
     GPIO.output(18, False) #Turn off LED

time.sleep(10)  # Delay for 10 seconds

dom-robinson, 2021-09-29:
Ill tag this in as an enhancement (for the 'to do' lists!) - if you wanted to contribute the code do please generate a working PR, but try to make sure it has error handling and some exposure in the settings.yml so it can be defaulted to off, but turned on where users have the requirement (default off means it won't break systems where GPIO is used for other things etc)...

colinluthier, 2022-01-28:
settings.yml should have the gpio pin and an enable setting as options.