jekhokie/raspberry-noaa-v2 #394: ERRORS -- Software fails to create tweepy configuration and fails to recognize twitter keys.. SOLVED (worked for me)

Dzewa, 2021-06-02:
The software keeps failing to create the necessary file for twitter functionality. I had to manually create the "**~/.tweepy.conf**" file. For those who are seeing "_ERROR: No Twitter consumer key defined_" be sure that your config file is written exactly like this:

**export TWITTER_CONSUMER_API_KEY=**"oKj3lh..."
**export TWITTER_CONSUMER_API_SECRET=**"Kug3..."
**export TWITTER_ACCESS_TOKEN=**"8346..."
**export TWITTER_ACCESS_TOKEN_SECRET=**"TMtu..."

(obviously add your own keys inside the quotes)

this solution worked for me and I now have twitter functionality.

Cadair, 2021-06-02:
Our documentation could probably be a little clearer, but it is expected that the `~/.tweepy.conf` file has to be created by the user.

Cadair, 2021-06-02:
I am going to close this in favour of #395 which is my preferred solution. Thanks for reporting :smile:

closed 2021-06-02 by Cadair