jekhokie/raspberry-noaa-v2 #399: [Error] software not installing meteor_demod

Dzewa, 2021-06-04:
Kept getting errors during meteor passes. Audio is captured and log shows passes upwards of 15 minutes long, but as the scripts call the demod process, it fails with the output "/usr/bin/meteor_demod: no such file or directory"

[image]

**Support Output**
Details about environment
=============================================
Current local date/time: Thu 03 Jun 2021 10:31:22 PM EDT
Current date/time (ms):  1622773882
Repo git hash:           8999e41666e32f91f276cce94fc1fdb0af4d388a
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.2
Architecture:            Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [69 Fri Jun  4 22:54:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210604-225447 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622861687 987 82 Northbound W
  * [66 Sat Jun  5 20:00:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210605-200018 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622937618 904 46 Northbound E
  * [67 Thu Jun  3 23:06:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210603-230644 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622776004 978 61 Northbound W
  * [74 Fri Jun  4 20:17:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210604-201753 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622852273 986 50 Northbound W
  * [70 Sat Jun  5 11:24:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210605-112432 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622906672 982 86 Southbound E
  * [78 Fri Jun  4 09:42:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210604-094233 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622814153 916 35 Southbound W
  * [81 Sat Jun  5 20:36:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210605-203655 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622939815 977 76 Northbound W
  * [75 Sat Jun  5 08:47:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210605-084747 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622897267 972 58 Southbound W
  * [80 Sat Jun  5 09:22:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210605-092227 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622899347 944 58 Southbound W
  * [73 Fri Jun  4 08:59:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210604-085938 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622811578 954 44 Southbound W
  * [68 Fri Jun  4 11:36:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210604-113627 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622820987 973 70 Southbound W
  * [71 Sat Jun  5 22:42:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210605-224254 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622947374 993 74 Northbound E
  * [79 Fri Jun  4 20:56:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210604-205659 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622854619 965 47 Northbound W
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: -6
  noaa_15_enable_bias_tee: false
  noaa_15_gain: 34
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 25
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: -6
  noaa_18_enable_bias_tee: false
  noaa_18_gain: 34
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 25
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: -6
  noaa_19_enable_bias_tee: false
  noaa_19_gain: 7.7
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 25
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: -6
  meteor_m2_enable_bias_tee: false
  meteor_m2_gain: 34
  meteor_m2_sun_min_elevation: 6
  meteor_m2_sat_min_elevation: 25
  meteor_m2_memory_threshold: 800
---------------------------------------------
USB Device Map:
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 004: ID 413c:8505 Dell Computer Corp.
  Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 29.6G  0 disk
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 29.3G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       30166572 4738620  24148968  17% /
  devtmpfs          823564       0    823564   0% /dev
  tmpfs             955660       0    955660   0% /dev/shm
  tmpfs             955660    8944    946716   1% /run
  tmpfs               5120       4      5116   1% /run/lock
  tmpfs             955660       0    955660   0% /sys/fs/cgroup
  tmpfs            1024000       0   1024000   0% /var/ramfs
  /dev/mmcblk0p1    258095   48790    209306  19% /boot
  tmpfs             191132       4    191128   1% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:        1911320 kB
  MemFree:          966088 kB
  MemAvailable:    1577300 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Elonics E4000 tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00000001
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

dom-robinson, 2021-06-07:
What does line 30 in your settings.yml file say?

Meteor_demod shouldnt be in master - thats in dev and still being worked on . If you want to get meteor_demod going check where we have got to here [https://github.com/jekhokie/raspberry-noaa-v2/issues/160](https://github.com/jekhokie/raspberry-noaa-v2/issues/160) - if you go down the thread there is a nearly-workable method that should get you as far as captures with bad map overlays, but its no where near ready for prime time :)

All hands welcome - there is a splinter chat in the discord where we sometimes get a few moments to progress it!

Dzewa, 2021-06-07:
line 30 in settings.yml specifies 'meteor_receiver: rtl_fm'
  
however, with a fresh install and after adjusting my settings.yml file, running the script **./install_and_upgrade.sh** was still checking to see if meteor_demod was installed and if not, was installing it by default. I was constantly getting errors after a meteor pass, saying that the process failed to produce an image because meteor_demod (which was _supposed_ to be located in /usr/bin) was not found.. 

After digging around, I decided to manually install the meteor_demod software but it kept being placed into usr/LOCAL/bin,  therefore the scripts couldn't find it. so I moved it from /usr/LOCAL/bin, to the proper place, being /usr/bin (where the scripts point to) and all is well. I am now receiving meteor images just fine. I had this discussion a few days ago with @Cadair and we pondered over the issue a bit, and the steps that I took fixed the issue for me.

overclock25, 2021-06-10:
> 
> 
> Kept getting errors during meteor passes. Audio is captured and log shows passes upwards of 15 minutes long, but as the scripts call the demod process, it fails with the output "/usr/bin/meteor_demod: no such file or directory"
> 
> [image]
> # **Support Output**
> Details about environment
> ## Current local date/time: Thu 03 Jun 2021 10:31:22 PM EDT
> Current date/time (ms): 1622773882
> Repo git hash: [8999e41](https://github.com/jekhokie/raspberry-noaa-v2/commit/8999e41666e32f91f276cce94fc1fdb0af4d388a)
> Raspberry Pi Model: Raspberry Pi 4 Model B Rev 1.2
> Architecture: Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
> Num CPUs: 4
> 
> 'at' Scheduled Jobs (Captures):
> 
>     * [69 Fri Jun  4 22:54:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210604-225447 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622861687 987 82 Northbound W
> 
>     * [66 Sat Jun  5 20:00:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20210605-200018 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622937618 904 46 Northbound E
> 
>     * [67 Thu Jun  3 23:06:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210603-230644 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622776004 978 61 Northbound W
> 
>     * [74 Fri Jun  4 20:17:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210604-201753 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622852273 986 50 Northbound W
> 
>     * [70 Sat Jun  5 11:24:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210605-112432 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622906672 982 86 Southbound E
> 
>     * [78 Fri Jun  4 09:42:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210604-094233 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622814153 916 35 Southbound W
> 
>     * [81 Sat Jun  5 20:36:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210605-203655 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622939815 977 76 Northbound W
> 
>     * [75 Sat Jun  5 08:47:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210605-084747 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622897267 972 58 Southbound W
> 
>     * [80 Sat Jun  5 09:22:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210605-092227 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622899347 944 58 Southbound W
> 
>     * [73 Fri Jun  4 08:59:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210604-085938 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622811578 954 44 Southbound W
> 
>     * [68 Fri Jun  4 11:36:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210604-113627 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622820987 973 70 Southbound W
> 
>     * [71 Sat Jun  5 22:42:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210605-224254 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622947374 993 74 Northbound E
> 
>     * [79 Fri Jun  4 20:56:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20210604-205659 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1622854619 965 47 Northbound W
> 
> 
> ## Satellite SDR Settings:
> noaa_15_schedule: true
> noaa_15_sdr_device_id: 0
> noaa_15_freq_offset: -6
> noaa_15_enable_bias_tee: false
> noaa_15_gain: 34
> noaa_15_sun_min_elevation: 6
> noaa_15_sat_min_elevation: 25
> noaa_18_schedule: true
> noaa_18_sdr_device_id: 0
> noaa_18_freq_offset: -6
> noaa_18_enable_bias_tee: false
> noaa_18_gain: 34
> noaa_18_sun_min_elevation: 6
> noaa_18_sat_min_elevation: 25
> noaa_19_schedule: true
> noaa_19_sdr_device_id: 0
> noaa_19_freq_offset: -6
> noaa_19_enable_bias_tee: false
> noaa_19_gain: 7.7
> noaa_19_sun_min_elevation: 6
> noaa_19_sat_min_elevation: 25
> meteor_m2_schedule: true
> meteor_m2_sdr_device_id: 0
> meteor_m2_freq_offset: -6
> meteor_m2_enable_bias_tee: false
> meteor_m2_gain: 34
[... 45 lines cut ...]
> Serial number: 00000001
> Serial number enabled: yes
> IR endpoint enabled: yes
> Remote wakeup enabled: no
> 
> Database tables:
> predict_passes =>
> CREATE TABLE predict_passes(
> sat_name text not null,
> pass_start timestamp primary key default (strftime('%s', 'now')) not null,
> pass_end timestamp default (strftime('%s', 'now')) not null,
> max_elev int not null,
[... cut ...]
> daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
> foreign key(pass_start) references passes(pass_start));

it seems that the file meteor_demod is not in usr/bin directory but in usr/local/bin.
I modified the path in the file common.sh with path /home/pi/raspberry-noaa-v2/scripts, replaced the string with the new correct path.
With this little change now I am able also to receive and decode Meteor M2 images.

JoshuaKimsey, 2021-08-29:
I have been having the same issue. I changed the path in the `common.sh` file from `/usr/bin` to `/usr/local/bin` and I'm waiting for the next Meteor pass, but I expect it to work as it should.

dom-robinson, 2021-09-29:
Cool: this is now a known bug. I am closing this issue thread since it is a repeat of #400

closed 2021-09-29 by dom-robinson