jekhokie/raspberry-noaa-v2 #400: Meteor 2

klikis, 2021-06-05:
i am not getting images from Meteor 2, this is my config file .. everything seems ok
[base station configurations.docx](https://github.com/jekhokie/raspberry-noaa-v2/files/6602218/base.station.configurations.docx)

BaumGuard, 2021-06-07:
I have the same problem here. Reception of NOAA satellites works fine, but I can't receive images from Meteor-M2.

overclock25, 2021-06-09:
Same same problem for me. NOAA  captures are good but about METEOR no images. Prediction, passes, polar diagrams are ok but no any picture at all. It might be an issue about the wider bandwidth needed for the reception of the russian satellite? Where to change it eventually?

dom-robinson, 2021-06-09:
can any one share any of the logging output?

overclock25, 2021-06-09:
Yes, that is the part of my log about the meteor process, it seems that it cannot find a file in order to demodulate the signal:

`09-06-2021 07:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : The system have enough space to store a Meteor pass on RAM
09-06-2021 07:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Free : 2886 ; Required : 600
09-06-2021 07:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true
09-06-2021 07:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Starting rtl_fm record
INFO : Recording at 137.1000 MHz...
09-06-2021 07:14 /home/pi/raspberry-noaa-v2/scripts/audio_processors/meteor_record_rtl_fm.sh INFO : Recording at 137.1000 MHz...
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 48.00 dB.
Tuner error set to -6 ppm.
Tuned to 137388000 Hz.
Oversampling input by: 4x.
Oversampling output by: 1x.
Buffer size: 7.11ms
Allocating 15 zero-copy buffers
Sampling at 1152000 S/s.
Output at 288000 Hz.
Signal caught, exiting!

User cancel, exiting...
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Demodulation in progress (QPSK)
*/home/pi/raspberry-noaa-v2/scripts/audio_processors/meteor_demodulate_qpsk.sh:** line 21: /usr/bin/meteor_demod: No such file or directory*
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing spectrogram
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing polar graph of direction for pass
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Free memory : 2617 ; Total RAMFS usage : 267M
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Deleting audio files
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Decoding in progress (QPSK to BMP)
Input file "/home/pi/raspberry-noaa-v2/tmp/meteor/METEOR-M-2-20210609-071417.qpsk" not found!
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh ERROR : Decoding failed, either a bad pass/low SNR or a software problem
09-06-2021 07:26 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Total processing time: 00:12.18
`

`09-06-2021 08:53 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : The system have enough space to store a Meteor pass on RAM
09-06-2021 08:53 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Free : 2834 ; Required : 600
09-06-2021 08:53 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true
09-06-2021 08:53 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Starting rtl_fm record
INFO : Recording at 137.1000 MHz...
09-06-2021 08:53 /home/pi/raspberry-noaa-v2/scripts/audio_processors/meteor_record_rtl_fm.sh INFO : Recording at 137.1000 MHz...
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 48.00 dB.
Tuner error set to -6 ppm.
Tuned to 137388000 Hz.
Oversampling input by: 4x.
Oversampling output by: 1x.
Buffer size: 7.11ms
Allocating 15 zero-copy buffers
Sampling at 1152000 S/s.
Output at 288000 Hz.
Signal caught, exiting!

User cancel, exiting...
09-06-2021 09:08 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Demodulation in progress (QPSK)
/home/pi/raspberry-noaa-v2/scripts/audio_processors/meteor_demodulate_qpsk.sh: line 21: /usr/bin/meteor_demod: No such file or directory
09-06-2021 09:08 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing spectrogram
09-06-2021 09:08 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing polar graph of direction for pass
09-06-2021 09:09 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Free memory : 2483 ; Total RAMFS usage : 348M
09-06-2021 09:09 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Deleting audio files
09-06-2021 09:09 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Decoding in progress (QPSK to BMP)
Input file "/home/pi/raspberry-noaa-v2/tmp/meteor/METEOR-M-2-20210609-085319.qpsk" not found!
09-06-2021 09:09 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh ERROR : Decoding failed, either a bad pass/low SNR or a software problem
09-06-2021 09:09 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Total processing time: 00:16.00
`

BaumGuard, 2021-06-09:
Could you please tell me where I can find the log file?

overclock25, 2021-06-09:
> 
> 
> Could you please tell me where I can find the log file?

launch this command: 
less /var/log/raspberry-noaa-v2/output.log

overclock25, 2021-06-09:
FOLLOW UP: 
it seems that the file meteor_demod is not in usr/bin directory but in usr/local/bin.
I modified the path in the file common.sh with path /home/pi/raspberry-noaa-v2/scripts, replaced the string with the new correct path then tomorrow morning I will see if it will work.

sjeff35, 2021-06-10:
> FOLLOW UP, IT WORKS:
> it seems that the file meteor_demod is not in usr/bin directory but in usr/local/bin.
> I modified the path in the file common.sh with path /home/pi/raspberry-noaa-v2/scripts, replaced the string with the new correct path.
> With this little change now I am able also to receive and decode Meteor M2 images.
> So check your .log file and IF you see the same error like mine, act as described here above.

Thanks for this!  I had the same issue.   Waiting on the next pass to see if it works!

BaumGuard, 2021-06-11:
Now it works for me as well. But can someone tell me where I can change the color settings for the Meteor images? Normally I use another program which produces images with intenser colors.

dom-robinson, 2021-06-12:
Well that is odd - meteor_demod is really still in dev - we are reccomending you use 'rtl_fm' or 'gnuradio' as the demodulation pathway for m2 at the moment. 

However its interesting that its broadly working (apart from the path setting) - could anyone of you who has already got that working share an output image? (Last time i tried to merge it in i spotted a map alignment image, but that may have been a local  clock thing, and i was waiting on others to try.... )

Assuming this is working then it may be time to promote meteor_demod to being a standard option in master.

Thanks @overclock25 ..

BaumGuard, 2021-06-12:
@dom-robinson 
Ok, I've attached two images for comparison. Both were received in the morning but on different days.

This one is from raspberry-noaa-v2:
[image]

And this one is from mlrpt which I use to receive images from Meteor M2:
[image]

In the config file of mlrpt there is a section for the color configuration.

dom-robinson, 2021-06-12:
Nice images - interesting differences :)

Neither are doing the map overlay which was where i got jammed up with https://github.com/jekhokie/raspberry-noaa-v2/issues/160 thinking it was a local issue.

OK _ so (not sure if you are on the [discord group](https://discord.gg/563QzaDP) - but there is a meteor_demod break out thread on there - just trying to nail why the ansible is installling meteor_demod into the /usr/local/bin folder when it seems to specify /usr/bin correctly...

if we can find out why its dropping into the wrong folder ill try to get that fix out to master asap. Be really nice to have that sorted at long last!

Thanks for the input!!

BaumGuard, 2021-06-12:
@dom-robinson 
Maybe you can add a function that shifts meteor_demod to /usr/bin if it isn't there already.
Could you also use mlrpt for the reception of Meteor M2?

Before I discovered this project, I wrote my own automation program for lrpt reception which I now use simultaneously with raspberry-noaa-v2. If you are interested: https://github.com/BaumGuard/AutoLRPT

dom-robinson, 2021-09-29:
Flagging this thread as a bug so we come back to it once the current PR release work is done elsewhere!

dom-robinson, 2021-09-29:
@Cadair I actually think this might be a quick ansible fix - might be a 10 min job for you (and an hour for me ;)

Cadair, 2021-10-05:
Wouldn't the simple fix here just be changing this line: https://github.com/jekhokie/raspberry-noaa-v2/blob/master/scripts/common.sh#L38 to be:

```
METEOR_DEMOD="$(which meteor_demod)"
```
?

ajcollins, 2021-10-07:
I experienced the same issue installing on my Raspberry Pi. I've added @Cadair's proposed change in #450.