jekhokie/raspberry-noaa-v2 #401: Self Signed SSL Certificates overwriting Purchased Certificate

5foot8, 2021-06-07:
**Please Describe the problem to be solved**

1. SSL Certificates - one of the main PC's I use to access my R-NOAA-v2 does not allow self-signed certificates.  I have manually added after much pain a trusted certificate which I purchased to secure my server.
2. The satvis.space "plugins" SSL certificate shows as not being from a trusted Certificate Authority and fails to load with a security error, so no pretty visualisation at the top of the page.

The problem is I cannot run the install and update script because it writes over my trusted certificate with a 'useless' self-signed one, and unpicking the self-signed certificate with the authentic one didn't go well on first three attempts.  

* Make it possible to run the install / update script with an option to leave SSL certificate in place.  I had to re-install my whole OS and r-noaa-v2 system
* Suggest that the satvis.space use a Certificate Authority other than LetsEncrypt which is not regarded with a high enough trust to get onto networks with strict security protocols. [ I realise this could very well be beyond your control]

If there are multiple solutions, please present each one separately.

dom-robinson, 2021-06-07:
HI -

Answer (ish) for Q1:   I'm not the authority on the SSL certs here - @Cadair might be worth tagging, but when @jekhokie wrote the core system and exposed the website with any certs he did make some security mention - i think there are some comments in the settings.yml about it that are relevant. Ultimately adding a 'cater for everyone's SSL setup' workflow was tricky from memory, and he opted for self-cert for ease / focus since it works in most cases.

At the end of the day its an NGINX webserver and if you have a specific requirement it shouldn't be too hard to modify with NGIX help online.. as it sounds like you have. I am a bit surprised you had to reinstall from OS to get it to work, but as i say I am not the authority on that.

Answer for Q2: As for  satvis i know that bit of the system quite well.

We iframe the satvis.space website into the page, and pass url queries from the satellite hyperlinks to the frame to fill the hours between the passes :) - that means that the satvis.space thing is entirely externally hosted at [https://satvis.space](https://satvis.space) - i have exchanged a few messages with him (about giving us a subset of just the Noaa 15,18 and 19 and MeteorM2 sats in a .tle which we can query instead of the full set he uses as default) but i think he is occupied with other things - that said occasionally his cert has failed and we had to wait a day or two for him to reset it.

Perhaps try asking him?

Let us know 👍

5foot8, 2021-06-08:
> HI -
>
> Answer (ish) for Q1: I'm not the authority on the SSL certs here - @Cadair might be worth tagging, but when @jekhokie wrote the core system and exposed the website with any certs he did make [...]

On the latest attempt before running the install and update script I made a copy of the `nginx/sites-available/default-tls` and once it's complete you then replace with the saved version.  This appears to be working for me.
> 
> At the end of the day its an NGINX webserver and if you have a specific requirement it shouldn't be too hard to modify with NGIX help online.. as it sounds like you have. I am a bit surprised you had to reinstall from OS to get it to work, but as i say I am not the authority on that.

The reinstall was due to the fact my original install was on a RPi with a fairly mature install, it had been on the machine for a good amount of time.  This reinstall allowed me to start afresh to make sure that the decisions I was making were correct.  It was this time that allowed me to rationally locate the file which was causing the challenge and not fall into the same trap again.

The simplest route I can see to the SSL certification challenge is if you have a certificate you tell it where it is located in the config file.  If no certificate route is supplied then a self-signed one is created.  It would obviously say that if you don't understand the question then leave alone.
> Answer for Q2: As for satvis i know that bit of the system quite well.
>
> We iframe the satvis.space website into the page, and pass url queries from the satellite hyperlinks to the frame to fill the [...]

Will do!
> 
> Let us know 👍

Sure.

Andrew.

dom-robinson, 2021-06-12:
> The simplest route I can see to the SSL certification challenge is if you have a certificate you tell it where it is located in the config file. If no certificate route is supplied then a self-signed one is created. It would obviously say in the config file that if you don't understand the question then leave alone.

Ah this does seem like completely good way to approach the issue. Pretty sure the Ansible setup can check for this, but i am going to tag in @Cadair since he is that bit stronger on the ansible side...

Thanks Andrew :)