jekhokie/raspberry-noaa-v2 #402: Auto configure Lat/Long/Alt for receiving site.

paradoxewan, 2021-06-09:
Have just purchased a cheap GPS receiver. This is doing good things for synchronising the clock on the PI.

For portable use of the software, would be great it this could update the configuration file for the site, and trigger a recalculation of the passes for the new location. Would take one configuration step away fro things.

If the configuration files are moving inside the GUI, even having a button that pulls the lat/long from the GPSD demon and can replace the values in the configuration file would be great.

5foot8, 2021-06-10:
> Have just purchased a cheap GPS receiver. This is doing good things for synchronising the clock on the PI.

@paradoxewan What method are you using for syncing the clock on the RPi?  

It's relatively difficult to get an accurate time without a network connection and NTP, unless you also have an addon physical Real Time Clock (RTC) module, RPi does not have a physical clock module.  The accuracy from the GPS is the Pulse Per Second (PPS) output that regulates the clock, not all GPS units have this output, especially the cheap ones.  The front of the pulse is accurate for the start of each second.  

If my understanding is correct (NTP) will not allow large changes of the clock and takes time to sync with the time source moving it slowly towards the correct time, unless connected via the network to a atomic clock or other stratum 1 time source.    A GPS principled time source is classed as a stratum 1 time source as it's only one step from the atomic clock, in the satellite.  I personally run an GPS principled NTP server on a spare RPi.  It's not unusual for my GPS PPS signal in NTP to be ignored in favour of the internet synced atomic clocks, even though technically its the most accurate time source in the house.  Once it's been running for a while and the system clock is approaching accurate then a lot of the time the PPS time source will become one of the primary time providers.  Setting the clock with a GPS is more difficult and involved than just adding a GPS receiver.  
> 
> For portable use of the software, would be great it this could update the configuration file for the site, and trigger a recalculation of the passes for the new location. Would take one configuration step away from things.

How often do you take your installation portable?  R-NOAA-V2 requires a network connection to download the weather satellites Kepler Elements. The only ongoing reason I could think of would be if you were in a boat or motor home but you still need internet for the data files.
> 
> If the configuration files are moving inside the GUI, even having a button that pulls the lat/long from the GPSD demon and can replace the values in the configuration file would be great.

Setting the location using GPSD is great in theory, in practice even the simple fact some GPS dongle units won't start unless they are plugged in when the machine is turned on could make for a very frustrating user experience.  KM4ACK in his Ham Radio Pi Installation wrote a software to attempt to force the RPi to connect to a GPS devices, and even then it doesn't always work.  Once again, it could work wonders and it could also cause great confusion and minor stress - setting your own location is far more easily understood, with many websites giving you very accurate co-ordinates by dropping pins on maps.

If we were having to drive pan and tilt antenna rotators I could see the need to have such an incredibly accurate location and time source.  With the stationary antennas mostly used for these satellites even if you're 15 seconds out and 10 km away with your location you will still get a recording/image. 

I apologise for the length of this response, I would not want users thinking that adding a cheap GPS module instantly makes your system location and time accurate.  Under the right conditions, with an optimised system, you may get close to these desires but in reality the likelihood of it not working out-of-the-box would be high.

colinluthier, 2021-07-12:
I also run a Stratum 1 server in my network. It can be made very accurate without internet as long as the leap seconds are up to date. While Keplers are needed, even a connection every few days would be enough. My Windows setup only downloaded Keplers once a week. 

The advantage to this idea is for a mobile receiving station.

paradoxewan, 2021-10-09:
> @paradoxewan What method are you using for syncing the clock on the RPi?
>
> It's relatively difficult to get an accurate time without a network connection and NTP, unless you also have an addon [...]

We are relying on having an NTP server (online connection). The suggestion for the GPS we to make it easy when moving local sites for the unit not negating the requirement for connection.