jekhokie/raspberry-noaa-v2 #406: Passes wont delete in mobile view

stevecrow74, 2021-06-18:
**Describe the bug**
on iPhone 7, in safari, in admin passes view ,  the delete confirmation box pops up, but the box is empty? it only seems to work on PC.
[image]

Raspberry pi 3B+
iPhone7

5foot8, 2021-06-21:
Definitely works in iPhone 12 browsers **Safari**,**Chrome**, **Firefox**, **DuckDuckGo**, **Edge** and even **Onion**. 
The only browser that failed was **Brave**, but it appeared to be security based as I'm running a full SSL certificate on my R-NOAA-2 website, and it appeared to get confused.

stevecrow74, 2021-06-21:
It just seems to be safari on iPhone 7. I can delete passes with chrome on the phone. So possibly problem my end. I’ll look into it this side and if I figure it out I’ll post back up a solution.

dom-robinson, 2021-08-25:
@stevecrow74  if its isolated ill close this for now - can reopen if you let us know its still and issue..

closed 2021-08-25 by dom-robinson