jekhokie/raspberry-noaa-v2 #407: no Flipping images METEOR

overclock25, 2021-06-24:
Hi,
I have noticed that images are not flipping. This happens only with METEOR images, with NOAA is all ok.
Any suggestions about where to check/verify?

Thanks

dom-robinson, 2021-06-24:
hmm - there was a recent PR merged that should have fixed it. How recent is your build?

[https://github.com/jekhokie/raspberry-noaa-v2/pull/398](https://github.com/jekhokie/raspberry-noaa-v2/pull/398)

overclock25, 2021-06-25:
> 
> 
> hmm - there was a recent PR merged that should have fixed it. How recent is your build?
> 
> #398

Hi Dom, 
I have corrected manually the script by adding some missing instructions, I guess I have an older version but now I have solved the problem and also Meteor images are correctly oriented. 
Thanks for having gave to me the reference of the issue.

Problem solved.

closed 2021-06-25 by overclock25