jekhokie/raspberry-noaa-v2 #410: Passes not showing on Webpanel

Dentonknifeworks, 2021-07-01:
New install
Passes show on pi using the  ATQ command
Passes don't show on the Webpanel 
Any suggestions?

dom-robinson, 2021-07-01:
Either reboot or wait until after midnight - it needs to 'get going' :) - either option will do that!

Dentonknifeworks, 2021-07-01:
Nope Reboot did not do it.[image]Like I said ATQ shows a schedule today. Put on the WebPanel no passes are showing.[image]

Dentonknifeworks, 2021-07-01:
Ok so I am getting images stored on the PI the web page don't seem to be updating.
Images are in the /var/www/wx/images

Dentonknifeworks, 2021-07-01:
Closing as I am doing a reinstall

closed 2021-07-01 by Dentonknifeworks