jekhokie/raspberry-noaa-v2 #414: [Error] receive_noaa.sh : select pass_id after insert in database

jp112sdl, 2021-07-10:
**Describe the bug**
Like in [scripts/receive_meteor.sh](https://github.com/jekhokie/raspberry-noaa-v2/blob/3b27e438ab6018cd89544df4f49b1c5507e38251/scripts/receive_meteor.sh)
the correct `pass_id` can be fetched after `INSERT OR REPLACE INTO decoded_passes...`

All push routines are using the wrong `pass_id` (from the previous pass):
https://github.com/jekhokie/raspberry-noaa-v2/blob/3b27e438ab6018cd89544df4f49b1c5507e38251/scripts/receive_noaa.sh#L284-L358

dom-robinson, 2021-07-19:
Hiya - I don't use slack with the RPI so not quite able to verify this, but on reading the code I think i see what you are highlighting. 

I'd like to tag in @colinluthier and @Cadair . 

Overall looking at the code something seems a little wrong to me. 

I think that in [https://github.com/jekhokie/raspberry-noaa-v2/blob/3b27e438ab6018cd89544df4f49b1c5507e38251/scripts/receive_meteor.sh](/scripts/receive_meteor.sh) that current lines 284  to 337 should be cut and pasted in between lines 275 and 276 so be moved into the same group that performs email and discord pushes. 

I can't recall who added the slack updater - but for sure as @jp112sdl highlights - as it is is gets (pass_id - 1)...

mihajlo2003petkovic, 2023-10-04:
Is this issue still active?

jp112sdl, 2023-10-04:
I think it is fixed.
As I can see, all notifications (slack, twitter, ...) are now done _after_ getting the actual `pass_id`.