jekhokie/raspberry-noaa-v2 #415: [Error]

NobrainDev, 2021-07-19:
**Describe the bug**
ok so i'm not sure if it's a bug but, my recording of noaa, every recording of noaa, is too loud and there's a strange buzzing sound when the signal of noaa is very clear and as a result the image doesn't get decoded well.

**Screenshots and/or logs**

**Support Output**
Paste the output of running the `./support.sh` script here.

**Server information (please complete the following information):**
 - OS: Raspbian lite
 - Python version latest installed with raspberry-noaa-v2
 - SDR device RTL-SDR-v3

**Additional context**
Add any other context about the problem here.

dom-robinson, 2021-07-19:
Hiya - so it sounds like a hardware issue, but it might be something that can be tweaking via software: What are your gain settings in the settings.yml?

If you are set to autogain ('0') then try to fix the gain - start around '42' and see if that helps.

if you get the buzzing when you are set very very low then the noise is likely to be being internally generated between the SDR and the PI. In that case it might be worth looking at possibly noise sources in the PI such as other USB / Hardware that is active on the PI (eg unnecessary bluetooth / wifi or badly shielded USB devices etc)..

Hope that helps

NobrainDev, 2021-07-19:
currently the gain is set to max for all NOAA sats, except for meteor which is slightly belove 49.
i didn't get this issue with rtl_tcp when streaming to my computer and recording noaa so i don't know if it could be noise.

dom-robinson, 2021-08-25:
I woudl bring the gain down to 42 or below if you can.

dom-robinson, 2021-09-29:
did that fix you?

dom-robinson, 2021-10-09:
closing - assuming all fixed now

closed 2021-10-09 by dom-robinson