jekhokie/raspberry-noaa-v2 #434: Automated File Transfer or remote storage solution for images and audio files.

RajSinghBains, 2021-08-25:
in cases where the raspberry pi is in a remote location such as a backyard shed, where they also have a limited built in storage soulution. having built-in functionality for NAS support or usb storage expantion

Impliment a currently working or system that allows the user to assign a url of either a ftp location or local storage location( such as drives, where it stores all of the data it gathers, such as audio files, images)

An additional feature to save configured settings and being able to revert back to it if the software was to crash, saves alot of time fiddling with settings manually.

dom-robinson, 2021-08-25:
Hey @RajSinghBains - so at the moment the variable that sets the storage location is hardcoded in the ansible setup here >>

[https://github.com/jekhokie/raspberry-noaa-v2/blob/91d13ac2fb43bf22751e45ac959c22fda7ee06b3/ansible/group_vars/all.yml#L15](https://github.com/jekhokie/raspberry-noaa-v2/blob/91d13ac2fb43bf22751e45ac959c22fda7ee06b3/ansible/group_vars/all.yml#L15)

```output_root: /srv```

If you were to manually change that before install you could map it to a USB drive or a NAS. Given that each USB or NAS will come online in a different way it would be difficult to say what you should change it to. 

I did have a little look at if it might make sense to move that config into settings.yml - however I am yet to really convince myself that letting anyone arbitrarily change that setting would not have unintended consequences. 

I don't have time this week to do a clean rebuild (but I can see that I soon will need to since there are a number of things backing up that it seems to be falling on me to solve!) 

If you feel brave then do try to make the change and let me know here what happens - that would be very useful. 
But if you don't feel ready to try that yourself then please wait until mid-september when I hope to find a few days to get a clean build with latest code in and get some of the backlog sorted. I would be keen to get FTP working too when i do that since i know a lot of folks would like it.

Equally if anyone else has time to do these things then Im happy to review PRs !! :)

dom-robinson, 2021-08-25:
Have created a Project 'TO DO' list and included these requests.

darkorb, 2022-01-17:
It's not FTP, but I've been lightly working on a branch that pushes images (for now) to an S3 compatible endpoint for long term storage. Would that be of any use? It _should_ work with any S3 provider (e.g. Minio, Wasabi, Backblaze etc.).

Fair warning, I'm not good at writing stuff so I mostly reused bits that already existed in the code base and just added as a post processor with the Ansible additions as needed but if it'd be of use to someone I'm willing to try getting it tided up/make changes to what I mangled together for my own use.

My use case for it was so I could include the images in my weather stations site without exposing the panels servers to the web at all fwiw, along side actually keeping them for a longer time.

RajSinghBains, 2022-01-19:
Thanks but my use case is a little specific and the hardware platform I use peanut support the use of s3, I am sure though that your implementation can be useful for others whom are using s3 for their use cases, so i am sure that it can be of some use to others if you with to go ahead with cleaning up and making it available for use.

darkorb, 2022-01-19:
If SCP would work for you I could quite likely adapt the S3 script to do a SCP push as long as you were using SSH Keys?

RajSinghBains, 2022-01-19:
Sorry needed a quick read up on SCP and it’s workings, and I don’t see Any immediate issues, but I am quite out of my field here really when It comes to file transfer over network with anything really. 

To expand a-bit on my criteria as I haven’t done so properly already, the transfer needs to be compatible with QNAP hardware/software and has to be internal to the network, as long as it meets those two points I see no real issue really.

Cadair, 2022-01-24:
My solution to this was to just make `/srv` a smb mount fwiw