jekhokie/raspberry-noaa-v2 #441: WINTER / SUMMER Meteor M2 sensor switching

dom-robinson, 2021-08-28:
@everyone to switch Meteor M2 sensors for winter you need to edit the file 

```scripts/image_processors/meteor_decode_bitstream.sh```

And switch the ```#``` comment between winter and summer mode as of today>>

```#  winter
$MEDET_ARM "${INPUT_BITSTREAM}" "${OUTPUT_BMP}" -cd -r 68 -g 65 -b 64 -na -s
#  summer
#$MEDET_ARM "${INPUT_BITSTREAM}" "${OUTPUT_BMP}" -r 66 -g 65 -b 64 -na -s```

Becomes

```#  winter
#$MEDET_ARM "${INPUT_BITSTREAM}" "${OUTPUT_BMP}" -cd -r 68 -g 65 -b 64 -na -s
#  summer
$MEDET_ARM "${INPUT_BITSTREAM}" "${OUTPUT_BMP}" -r 66 -g 65 -b 64 -na -s```

Hope that helps 🙂

```

dom-robinson, 2021-08-29:
This is still being debugged!!!

this will only apply if you are using gnuradio mode an there may be more tweaks needed

JoshuaKimsey, 2021-08-29:
Also, Meteor seems to be operating in it's normal 1-2-3 Summer mode now, so this shouldn't be necessary

closed 2021-09-29 by dom-robinson