jekhokie/raspberry-noaa-v2 #448: [Error] sun.py ValueError: invalid literal for int() with base 10: ''

fg8oj, 2021-09-14:
Hi, getting this error : 

/home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210914-233934 /home/pi/raspberry-noaa-v2/tmp/orbit.tle 
expr: non-integer argument
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa-v2/scripts/tools/sun.py", line 16, in <module>
    date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(int(sys.argv[1])-(timezone*60*60)))
ValueError: invalid literal for int() with base 10: ''
INFO : Starting rtl_fm record
INFO : Producing spectrogram
INFO : Producing polar graph of direction for pass
INFO : Bulding pass map
/home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh: line 178: [: : integer expression expected
INFO : Decoding image
INFO : Decoding image
INFO : Decoding image
INFO : Decoding image
INFO : Decoding image
INFO : Decoding image
rm: cannot remove '/home/pi/raspberry-noaa-v2/tmp/map/NOAA-19-20210914-233934-map.png': No such file or directory
INFO : Total processing time: 00:00.04

Raspberry Pi 3B
Linux noaa 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux

=============================================
Details about environment
=============================================
Current local date/time: Tue Sep 14 13:37:47 BST 2021
Current date/time (ms):  1631623067
Repo git hash:           9cff24b5537ec9cf1cca83abdf8d2ed56626d69d
Raspberry Pi Model:      Raspberry Pi 3 Model B Rev 1.2
Architecture:            Linux noaa 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [2	Tue Sep 14 23:39:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210914-233934 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1631659174 975 63 Northbound E
  * [1	Tue Sep 14 14:49:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210914-144951 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1631627391 993 63 Southbound E
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: -6
  noaa_15_enable_bias_tee: true
  noaa_15_gain: 7.7
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: -6
  noaa_18_enable_bias_tee: true
  noaa_18_gain: 8.7
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: -6
  noaa_19_enable_bias_tee: true
  noaa_19_gain: 7.7
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: -6
  meteor_m2_enable_bias_tee: true
  meteor_m2_gain: 7.7
  meteor_m2_sun_min_elevation: 6
  meteor_m2_sat_min_elevation: 30
  meteor_m2_memory_threshold: 600
---------------------------------------------
USB Device Map:
  Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
  Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 28.9G  0 disk 
  |-mmcblk0p1 179:1    0  256M  0 part /boot
  `-mmcblk0p2 179:2    0 28.6G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       29519740 3665024  24622100  13% /
  devtmpfs          439916       0    439916   0% /dev
  tmpfs             473196       0    473196   0% /dev/shm
  tmpfs             473196   12292    460904   3% /run
  tmpfs               5120       4      5116   1% /run/lock
  tmpfs             473196       0    473196   0% /sys/fs/cgroup
  /dev/mmcblk0p1    258095   49209    208887  20% /boot
  tmpfs              94636       0     94636   0% /run/user/1000
  tmpfs              94636       0     94636   0% /run/user/0
  tmpfs            1024000       0   1024000   0% /var/ramfs
---------------------------------------------
Memory Info:
  MemTotal:         946392 kB
  MemFree:          149976 kB
  MemAvailable:     820576 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Failed to open rtlsdr device #0.
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

closed 2021-09-14 by fg8oj

dom-robinson, 2021-09-14:
i saw you closed this? Perhaps you already solved it, but 

```Using device 0: Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Failed to open rtlsdr device #0.```

that bit implies you are having some issues with the SDR card not being accessible?

Did you do a clean install on a new buster OS setup? Or something else? It might be that...

fg8oj, 2021-09-14:
Hi Dom, 
I was a clean install from this image : 2021-05-07-raspios-buster-armhf-lite.img
Nothing change, I really don't know why it didn't work last time.

=============================================
Details about environment
=============================================
Current local date/time: Tue 14 Sep 08:55:25 AST 2021
Current date/time (ms):  1631624125
Repo git hash:           9cff24b5537ec9cf1cca83abdf8d2ed56626d69d
Raspberry Pi Model:      Raspberry Pi 3 Model B Rev 1.2
Architecture:            Linux noaa 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [2	Tue Sep 14 18:39:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20210914-233934 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1631659174 975 63 Northbound E
  * [1	Tue Sep 14 09:49:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20210914-144951 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1631627391 993 63 Southbound E
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: -6
  noaa_15_enable_bias_tee: true
  noaa_15_gain: 7.7
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: -6
  noaa_18_enable_bias_tee: true
  noaa_18_gain: 8.7
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: -6
  noaa_19_enable_bias_tee: true
  noaa_19_gain: 7.7
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: -6
  meteor_m2_enable_bias_tee: true
  meteor_m2_gain: 7.7
  meteor_m2_sun_min_elevation: 6
  meteor_m2_sat_min_elevation: 30
  meteor_m2_memory_threshold: 600
---------------------------------------------
USB Device Map:
  Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
  Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 28.9G  0 disk 
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 28.6G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       29519740 3647024  24640100  13% /
  devtmpfs          439400       0    439400   0% /dev
  tmpfs             472680       0    472680   0% /dev/shm
  tmpfs             472680   12244    460436   3% /run
  tmpfs               5120       4      5116   1% /run/lock
  tmpfs             472680       0    472680   0% /sys/fs/cgroup
  tmpfs            1024000       0   1024000   0% /var/ramfs
  /dev/mmcblk0p1    258095   49209    208887  20% /boot
  tmpfs              94536       0     94536   0% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:         945364 kB
  MemFree:          760932 kB
  MemAvailable:     825828 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

dom-robinson, 2021-09-14:
Those logs look a bit better now : The SDR is recognised right?

I can’t see any issue with the settings - although you might want to boost your gains, here’s mine for ref >>

```
noaa_15_schedule: true
noaa_15_sdr_device_id: 0
noaa_15_freq_offset: -0
noaa_15_enable_bias_tee: false
noaa_15_gain: 32.8
noaa_15_sun_min_elevation: 6
noaa_15_sat_min_elevation: 30

noaa_18_schedule: true
noaa_18_sdr_device_id: 0
noaa_18_freq_offset: -0
noaa_18_enable_bias_tee: false
noaa_18_gain: 32.8
noaa_18_sun_min_elevation: 6
noaa_18_sat_min_elevation: 30

noaa_19_schedule: true
noaa_19_sdr_device_id: 0
noaa_19_freq_offset: -0
noaa_19_enable_bias_tee: false
noaa_19_gain: 32.8
noaa_19_sun_min_elevation: 6
noaa_19_sat_min_elevation: 30

meteor_m2_schedule: true
meteor_m2_sdr_device_id: 0
meteor_m2_freq_offset: -0
meteor_m2_enable_bias_tee: false
meteor_m2_gain: 25.4
meteor_m2_sun_min_elevation: 10
meteor_m2_sat_min_elevation: 30
```

fg8oj, 2021-09-14:
[image]
Nothing interesting decoded on a 63° elevation pass but at least process worked :)