jekhokie/raspberry-noaa-v2 #449: Should I be getting processed images for colour automatically ?

m1ctk, 2021-09-26:
I thought colour was part of the automation but it seems not ? I have t0o manually process with wxtoimg ?

dom-robinson, 2021-09-27:
should be automatic. can you share more details?

m1ctk, 2021-09-28:
> should be automatic. can you share more details?

What details do you want ? Simply wxtoimg is not processing for colour for some reason with me

ve3elb, 2021-09-28:
Share some details. Are you using your own build or RaspiNOAA Disk image. And use everything is automatic. ￼

m1ctk, 2021-09-28:
Noting like that. All I get in the passes tab is Meteor M2. Other Noaa images I have to find in the srv folder and are all black and white. Nothing is obvious to me in settings.yml as to why all images are only black and white. it seems wztoimg is no processing in to colour for somr reason - I am using the Buster 6 image from VL3LNB

ve3elb, 2021-09-28:
Best to just start over install a fresh copy of the disk image and go from there.

m1ctk, 2021-09-28:
> Best to just start over install a fresh copy of the disk image and go from there.

Tried that makes no difference

ve3elb, 2021-09-28:
Well in that case you are not doing part of the setup correctly. That could be a thing as to location, time setting, etc.

m1ctk, 2021-09-29:
> Well in that case you are not doing part of the setup correctly. That could be anything as to location, time setting, etc.

Time and date is correct as well as location. Paases show exactly the same tines as other sources like orbitron on my pc or the iss detector app on my phone

dom-robinson, 2021-09-29:
Can you share your settings.yml and the /var/log/raspberry-noaa-v2/output.log file please.

m1ctk, 2021-09-29:
---
# base station configurations
#   latitude: south values are negative
#   longitude: west values are negative
latitude: 50.86
longitude: -0.15

# time zone offset from UTC (for example, '-5' for US Eastern)
timezone_offset: +1

# ntp configurations
#   ntp_server - if you have a local server (e.g. stratum1), you can use this setting
#        to specify a hostname or ip address to communicate with the ntp server
ntp_server: ''

# test settings when running test scripts
#   test_gain - gain to use for scanner test scripts
#   test_sdr_device_id - device ID of the SDR device to be used for scanner test scripts
#   test_enable_bias_tee - whether to enable bias tee when running scanner test scripts
#   test_freq_offset - receiver frequency offset (PPM)
test_gain: 44.5
test_sdr_device_id: 0
test_enable_bias_tee: false
test_freq_offset: 0

# receiver settings
#   meteor_receiver - which receiver method to use (either 'rtl_fm' or 'gnuradio')
#     **WARNING**: 'gnuradio' does not work with certain SDR devices (e.g. it will not currently
#                  work with a RTL-SDR v3 dongle, as no image will be decoded from the bitstream)
meteor_receiver: 'rtl_fm'

# whether to schedule specific orbiting objects for capture
#   <satellite_name>_schedule - whether to schedule captures for the satellite
#   <satellite_name>_sdr_device_id - device ID of the SDR device to be used for recording for the satellite
#   <satellite_name>_freq_offset - receiver frequency offset (PPM) for the satellite capture
#   <satellite_name>_enable_bias_tee - whether to enable bias tee for the recording of the satellite
#   <satellite_name>_gain - gain setting for specific satellite captures
#   <satellite_name>_sun_min_elevation - threshold for sun elevation for specific satellite captures
#   <satellite_name>_sat_min_elevation - threshold for sat elevation for specific satellite captures
#   <satellite_name>_memory_threshold - for METEOR satellite, minimum free memory (MB) required to store pass in RAM

noaa_15_schedule: true
noaa_15_sdr_device_id: 0
noaa_15_freq_offset: 0
noaa_15_enable_bias_tee: false
noaa_15_gain: 44.5
noaa_15_sun_min_elevation: 0
noaa_15_sat_min_elevation: 20

noaa_18_schedule: true
noaa_18_sdr_device_id: 0
noaa_18_freq_offset: 0
noaa_18_enable_bias_tee: false
noaa_18_gain: 44.5
noaa_18_sun_min_elevation: 0
noaa_18_sat_min_elevation: 20

noaa_19_schedule: true
noaa_19_sdr_device_id: 0
noaa_19_freq_offset: 0
noaa_19_enable_bias_tee: false
noaa_19_gain: 44.5
noaa_19_sun_min_elevation: 0
noaa_19_sat_min_elevation: 20

meteor_m2_schedule: true
meteor_m2_sdr_device_id: 0
meteor_m2_freq_offset: 0
meteor_m2_enable_bias_tee: false
meteor_m2_gain: 44.5
[... 171 lines cut ...]

# push processing settings for sending images elsewhere
#   * NOTE: Make sure you set up your ~/.msmtprc file before enabling email push!
#   enable_email_push - whether to send all images to an external email
#   email_push_address - if enabled, address to send all images to
#   enable_email_schedule_push - whether to email an image of the nightly-created
#       pass-list schedule to the email destination
#   enable_discord_push - whether to push images to a Discord channel
#   discord_webhook_url - webhook url for the Discord channel
#   enable_twitter_push - whether to push images to a Twitter feed
#     * see docs/twitter_push.md for instructions
#   enable_matrix_push - whether to push images to a Matrix room
#     * see docs/matrix_push.md for instructions
enable_email_push: false
email_push_address: test@ifttt.com
enable_email_schedule_push: false
enable_discord_push: true
discord_webhook_url: ''
enable_twitter_push: false
enable_slack_push: false
slack_push_url: ''
slack_push_to: ''
slack_link: 'https://XXXX/captures/listImages'
enable_matrix_push: false
...

m1ctk, 2021-09-29:
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Scheduler resolved TLE endpoint in 0 seconds
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Downloading new TLE files from source
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2021-09-29 00:01:02--  http://www.celestrak.com/NORAD/elements/weather.txt
Resolving www.celestrak.com (www.celestrak.com)... 104.168.149.178
Connecting to www.celestrak.com (www.celestrak.com)|104.168.149.178|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 8904 (8.7K) [text/plain]
Saving to: '/home/pi/raspberry-noaa-v2/tmp/weather.txt'

     0K ........                                              100% 66.3K=0.1s

2021-09-29 00:01:02 (66.3 KB/s) - '/home/pi/raspberry-noaa-v2/tmp/weather.txt' saved [8904/8904]

FINISHED --2021-09-29 00:01:02--
Total wall clock time: 0.5s
Downloaded: 1 files, 8.7K in 0.1s (66.3 KB/s)
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2021-09-29 00:01:02--  http://www.celestrak.com/NORAD/elements/amateur.txt
Resolving www.celestrak.com (www.celestrak.com)... 104.168.149.178
Connecting to www.celestrak.com (www.celestrak.com)|104.168.149.178|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14784 (14K) [text/plain]
Saving to: '/home/pi/raspberry-noaa-v2/tmp/amateur.txt'

     0K .......... ....                                       100% 13.3K=1.1s

2021-09-29 00:01:04 (13.3 KB/s) - '/home/pi/raspberry-noaa-v2/tmp/amateur.txt' saved [14784/14784]

FINISHED --2021-09-29 00:01:04--
Total wall clock time: 1.5s
Downloaded: 1 files, 14K in 1.1s (13.3 KB/s)
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Clearing and re-creating TLE file with latest...
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Determining latest scheduled capture job date...
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Scheduling starting at 1633215659 ms through 1633301999 ms...
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Scheduling new capture jobs...
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Scheduling NOAA 15 captures...
/home/pi/.noaa-v2.conf: line 81: unexpected EOF while looking for matching `''
/home/pi/.noaa-v2.conf: line 93: syntax error: unexpected end of file
INFO : Scheduling capture for: NOAA-15 20211003-080920 35
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-15 20211003-080920 35
INFO : Scheduled capture with job id: 835
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduled capture with job id: 835
INFO : Scheduling capture for: NOAA-15 20211003-094911 41
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-15 20211003-094911 41
INFO : Scheduled capture with job id: 836
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduled capture with job id: 836
INFO : Scheduling capture for: NOAA-15 20211003-193425 87
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-15 20211003-193425 87
INFO : Scheduled capture with job id: 837
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduled capture with job id: 837
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule.sh INFO : Scheduling NOAA 18 captures...
/home/pi/.noaa-v2.conf: line 81: unexpected EOF while looking for matching `''
/home/pi/.noaa-v2.conf: line 93: syntax error: unexpected end of file
INFO : Scheduling capture for: NOAA-18 20211003-105025 43
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-18 20211003-105025 43
INFO : Scheduled capture with job id: 838
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduled capture with job id: 838
INFO : Scheduling capture for: NOAA-18 20211003-123115 36
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-18 20211003-123115 36
INFO : Scheduled capture with job id: 839
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduled capture with job id: 839
INFO : Scheduling capture for: NOAA-18 20211003-204232 24
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduling capture for: NOAA-18 20211003-204232 24
INFO : Scheduled capture with job id: 840
29-09-2021 00:01 /home/pi/raspberry-noaa-v2/scripts/schedule_captures.sh INFO : Scheduled capture with job id: 840
[... 185 lines cut ...]
..29-09-2021 09:17 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Total processing time: 00:00.20
29-09-2021 09:50 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Starting rtl_fm record
/home/pi/.noaa-v2.conf: line 81: unexpected EOF while looking for matching `''
/home/pi/.noaa-v2.conf: line 93: syntax error: unexpected end of file
INFO : Recording at 137.6200 MHz...
29-09-2021 09:50 /home/pi/raspberry-noaa-v2/scripts/audio_processors/noaa_record.sh INFO : Recording at 137.6200 MHz...
No supported devices found.
29-09-2021 09:50 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Producing pristine image
[... cut ...]
29-09-2021 09:50 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Producing polar graph of direction for pass
/home/pi/.noaa-v2.conf: line 81: unexpected EOF while looking for matching `''
/home/pi/.noaa-v2.conf: line 93: syntax error: unexpected end of file
29-09-2021 09:50 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Bulding pass map
Satellite: NOAA 15
Pass Start: 2021-09-29 08:50:13 UTC
Pass Duration: 14:44
Elevation: 40
Azimuth: 296
Direction: southbound
..29-09-2021 09:50 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Total processing time: 00:00.20

dom-robinson, 2021-09-29:
Ha - you are near me :) - im in Brighton too!

I think the error is simple ```ground_station_location: Brighton UK'```
you need an extra ' 

should be ```ground_station_location: 'Brighton UK'```

dom-robinson, 2021-09-29:
That missing '`' will mean that something that should be passed as a string might be causing a mess in the scheduling and on a cursory look it looks in the log file that the noaa schedulign is moaning - so hopefully its that simple...

m1ctk, 2021-09-29:
> Ha - you are near me :) - im in Brighton too!
> 
> I think the error is simple `ground_station_location: Brighton UK'` you need an extra '
> 
> should be `ground_station_location: 'Brighton UK'`

OK  but surely that wouldn't affect me getting colour not not ?!

m1ctk, 2021-09-29:
Good news t's made all the difference.  Getting all the  Enhancements now, Brilliant thanks to Dom !

dom-robinson, 2021-09-30:
Cracking long pass! Good. Closing :)

closed 2021-09-30 by dom-robinson