jekhokie/raspberry-noaa-v2 #460: [Error] No Image with noaa raspberry pi but works when connected to computer.

betolley, 2021-11-03:
I can capture NOAA images using the same antenna you can see in the first image.  I have tried to use my raspberry pi and I get what the other images have basically static. The antenna hasn't moved.   I can tune to local radio stations and hear them but get nothing with NOAA images.  I have tried using gain of 0, 20, 40 and 47 which show as supported with rlt_test.  Any ideas?
[image]
[image]
[image]
[image]
[image]

dom-robinson, 2021-11-04:
It does look like you have some very faint signal there, but clearly not strong enough for the decode to kick in. Good news is the rest of the RN2 setup seems to be working.

Perhaps try switching the mode between rtl_fm and gnuradio. (which SDR are you using by the way?)

Can you share any of the logging? See [https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/troubleshooting.md](https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/troubleshooting.md) for details (and a few other tips).

betolley, 2021-11-05:
Here is the log:

https://pastebin.com/8w508fGD

dom-robinson, 2021-11-12:
```usb_claim_interface error -6

FATAL: Failed to open rtlsdr device.```

This implies something else is using the SDR or its simply failing for some reason....

colinluthier, 2021-11-14:
The troubleshooting doc has some steps to listen to the radio. Make sure you can hear an fm station on the pi.

betolley, 2021-11-14:
I do and if I use airspy I can tune to the station in sdr# and it sounds
great.

dom-robinson, 2021-11-15:
have you tried with both modes? rtl_fm and gnuradio?

betolley, 2021-11-20:
Thanks that was it.  I switched to rtl_fm and then lowered the gain to 39 and now it works!

closed 2021-11-20 by betolley