jekhokie/raspberry-noaa-v2 #462: Consider options to remove use of "predict"

Cadair, 2021-11-04:
The "predict" satellite prediction program is not packaged for debian after stretch (the previous stable version).

We should consider either a) building predict manually or b) using other options like pypredict etc. This issue is to collect suggestions and work out how much effort this is going to be.

See https://github.com/jekhokie/raspberry-noaa-v2/pull/461#discussion_r742719849 for context.

dom-robinson, 2021-11-04:
good idea. pypredict looks like a good starting point.

I guess a logical start point will be to work out what needs 'mapping' from the current predict usage into pypredict - without a deeper play I am assuming much of the same input data (location, time, elevation mins etc, satellite of choice and location of TLE files etc) will be required, meaning we may just need pypredict installed by ansible and then a version  of [https://github.com/jekhokie/raspberry-noaa-v2/blob/master/scripts/schedule_captures.sh](https://github.com/jekhokie/raspberry-noaa-v2/blob/master/scripts/schedule_captures.sh) to cope with any nuances between them?

rhaamo, 2021-11-04:
you can find some code I used to predict passes with pypredict here https://github.com/rhaamo/autowx/blob/master/pypredict.py

and yes you only need lat, lon, min elevation, altitude and TLE

colinluthier, 2021-11-11:
The latest push has the altitude variable in settings.

colinluthier, 2022-01-28:
pypredict imports predict in its first line.

colinluthier, 2022-02-20:
I am back to looking at this. The archive of pypredict I was looking at was not the official pypredict. Here are the requirements:

The following are the current dependencies:

cartopy 0.18 or later.
cython 0.28 or later.
geos 0.2.2 or later.
matplotlib 3.2.1 or later.
numpy 1.18.2 or later.
Pillow 7.1.1 or later.
pykdtree 1.3.1 or later.
pymongo 3.10.1 or later.
PyQt5 5.14 or later.
pyshp 1.1.4 or later.
scipy 1.4.1 or later.
sgp4 2.7 or later.
shapely 1.5.6 or later.

Changing the schedule code is straightforward.