jekhokie/raspberry-noaa-v2 #463: dpkg: error processing archive wxtoimg-armhf-2.11.2-beta.deb (--install):  package architecture (armhf) does not match system (arm64)

benb0jangles, 2021-11-05:
During install i get error:

`dpkg: error processing archive wxtoimg-armhf-2.11.2-beta.deb (--install):  package architecture (armhf) does not match system (arm64)`

wxtoimg-armhf-2.11.2-beta.deb

Please any ideas? Pi4 is arm64 or armhf?

Cadair, 2021-11-05:
I assume you are running a pi4 with a 64bit OS?

benb0jangles, 2021-11-05:
> I assume you are running a pi4 with a 64bit OS?

yes correct

rhaamo, 2021-11-05:
https://unix.stackexchange.com/questions/625576/how-to-run-32-bit-armhf-binaries-on-64-bit-arm64-debian-os-on-raspberry-pi there might be a way to install and run armhf packages on arm64, however I can't test I only have pi3

Cadair, 2021-11-05:
I don't have a spare pi4 to test this with until next week either. Thanks for the bug report @benb0jangles

benb0jangles, 2021-11-05:
ok thanks

dinoj-bg, 2022-02-04:
To use wxtoimg on 64-bit architecture You need manual install 32-bit library.
just run apt update 
then apt install  libxft2:armhf libasound2:armhf 
Also You need to install manualy install wxtoimg with dpkg -i wxtoimg-armhf-2.11.2-beta.deb

Cadair, 2022-02-04:
If anyone feels motivated to figure out how to get ansible to do all those things conditionally based on architecture that would be a fantastic pull request.

Cadair, 2022-05-15:
If anyone fancies adding a note to the docs making it clear to people they need to use a 32bit image until we work through all the issues with the 64bit one that would be a great help.

dom-robinson, 2022-05-16:
[image]

Temp fix... lmk if we need to tweak..