jekhokie/raspberry-noaa-v2 #475: [Error] while installation

Eric738, 2021-11-28:
Good Afternoon

at the end of install i had 👍 
---------
[WARNING]: Updating cache and auto-installing missing dependency: python-apt
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "apt-get install --no-install-recommends python-apt -y -q", "msg": "E: Package 'python-apt' has no installation candidate", "rc": 100, "stderr": "E: Package 'python-apt' has no installation candidate\n", "stderr_lines": ["E: Package 'python-apt' has no installation candidate"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nPackage python-apt is not available, but is referred to by another package.\nThis may mean that the package is missing, has been obsoleted, or\nis only available from another source\nHowever the following packages replace it:\n  python3-apt-dbg python3-apt python-apt-doc python-apt-common\n\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Package python-apt is not available, but is referred to by another package.", "This may mean that the package is missing, has been obsoleted, or", "is only available from another source", "However the following packages replace it:", "  python3-apt-dbg python3-apt python-apt-doc python-apt-common", ""]}

PLAY RECAP *********************************************************************
localhost                  : ok=14   changed=2    unreachable=0    failed=1    skipped=3    rescued=0    ignored=0 

on raspbian bulleyes + raspi 4
key rtlsdr 
how i can fix it ?

thank for info
Eric

dom-robinson, 2021-11-29:
HEy Eric - there is an update being worked on for the bullseye OS [https://github.com/jekhokie/raspberry-noaa-v2/pull/468](https://github.com/jekhokie/raspberry-noaa-v2/pull/468)

It looks like it was merged in over the last few days so *should* be the master - however I think someone else has reported a similar issue - i Tagged @colinluthier in since he was more across this update than I am.  I am pretty sure the master is currently meant to be good for bullseye but no 100% sure having not tested on buster or bullseye myself. Hopefully Colin can help?

colinluthier, 2021-11-29:
@Eric738 Can you let me know what OS was installed? There is a known issue on the 64 bit Raspian OS.

Eric738, 2021-12-01:
ge
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
-----
Linux raspberrypi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
----
armv7l
32 bits

Eric

RyLeeRyno, 2021-12-11:
sudo apt install python-is-python3

If that doesn't work then update the dependencies.yml as below.

~/raspberry-noaa-v2/ansible/roles/common/tasks $ cat dependencies.yml
---
- name: install dependencies
  become: yes
  apt:
    update_cache: yes
    state: present
    name:
      - at
      - bc
      - cmake
      - ffmpeg
      - gmic
      - gnuradio
      - gr-osmosdr
      - imagemagick
      - jq
      - libatlas-base-dev
      - libjpeg9
      - libjpeg9-dev
      - libncurses5-dev
      - libncursesw5-dev
      - libsox-fmt-mp3
      - libusb-1.0-0
      - libusb-1.0-0-dev
      - libxft-dev
      - libxft2
      - predict
      - python3-pip
      - python-setuptools
      - socat
      - sox
      - sqlite3

Eric738, 2021-12-12:
thank info  i test

Eric