jekhokie/raspberry-noaa-v2 #479: Upon Install, I get the following Error

KE8AFF, 2021-12-12:
**Describe the bug**TASK [webserver : install php] **********************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'php7.4-fpm' is available"}
        to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP ******************************************************************************************************************************************************************************************
localhost                  : ok=47   changed=0    unreachable=0    failed=1

Error:   Something failed with the install - please inspect the logs above

A clear and concise description of what the bug is.

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well.

KE8AFF, 2021-12-13:
pi@raspberrypi:~/raspberry-noaa-v2 $ ./install_and_upgrade.sh
 * Checking for python3-pip...
dpkg-query: no packages found matching python3-pip
 *   python3-pip not yet installed - installing...
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dh-python gir1.2-glib-2.0 libexpat1-dev libgirepository-1.0-1 libpython3-dev libpython3.7-dev python-pip-whl python3-asn1
  python3-dbus python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring python3-keyrings.alt python3-li
  python3.7-dev
Suggested packages:
  python-crypto-doc python-cryptography-doc python3-cryptography-vectors python-dbus-doc python3-dbus-dbg gnome-keyring lib
  python-setuptools-doc
The following NEW packages will be installed:
  dh-python gir1.2-glib-2.0 libexpat1-dev libgirepository-1.0-1 libpython3-dev libpython3.7-dev python-pip-whl python3-asn1
  python3-dbus python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring python3-keyrings.alt python3-li
  python3-xdg python3.7-dev
0 upgraded, 25 newly installed, 0 to remove and 0 not upgraded.
Need to get 51.4 MB of archives.
After this operation, 84.3 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf python-pip-whl all 18.1-5+rpt1 [1,591 kB]
Get:2 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pytho                                           n3-lib2to3 all 3.7.3-1 [76.7 kB]
0% [2 python3-lib2to3 972 B/76.7 kB 1%] [Waiting for headers] [1 python-pip-whl                                            Get:3 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pytho                                           n3-distutils all 3.7.3-1 [142 kB]
Get:4 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf dh-py                                           thon all 3.20190308 [99.3 kB]
Get:5 http://archive.raspberrypi.org/debian buster/main armhf python3-pip all 18                                           .1-5+rpt1 [171 kB]
Get:6 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf libgi                                           repository-1.0-1 armhf 1.58.3-2 [78.8 kB]
Get:7 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf gir1.                                           2-glib-2.0 armhf 1.58.3-2 [143 kB]
Get:8 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf libex                                           pat1-dev armhf 2.2.6-2+deb10u1 [127 kB]
Get:9 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf libpy                                           thon3.7-dev armhf 3.7.3-2+deb10u3 [47.2 MB]
Get:17 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3-dev ar                                           mhf 3.7.3-1 [1,264 B]
Get:10 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf libp                                           ython3-dev armhf 3.7.3-1 [20.1 kB]
Get:11 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-asn1crypto all 0.24.0-1 [78.2 kB]
Get:12 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-cffi-backend armhf 1.12.2-1 [70.1 kB]
Get:13 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-crypto armhf 2.6.1-9+b1 [251 kB]
Get:14 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-cryptography armhf 2.6.1-3+deb10u2 [184 kB]
Get:15 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-dbus armhf 1.2.8-3 [95.9 kB]
Get:16 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3.7-dev armhf 3.7.3-2+deb10u3 [513 kB]
Get:18 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-entrypoints all 0.3-1 [5,508 B]
Get:19 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-gi armhf 3.30.4-1 [161 kB]
Get:20 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-secretstorage all 2.3.1-2 [14.2 kB]
Get:21 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-keyring all 17.1.1-1 [43.1 kB]
Get:22 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-keyrings.alt all 3.1.1-1 [18.2 kB]
Get:23 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-setuptools all 40.8.0-1 [306 kB]
Get:24 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-wheel all 0.32.3-2 [19.4 kB]
Get:25 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf pyth                                           on3-xdg all 0.25-5 [35.9 kB]
Fetched 51.4 MB in 26s (1,988 kB/s)
Selecting previously unselected package python3-lib2to3.
(Reading database ... 96675 files and directories currently installed.)
[... cut ...]
TASK [webserver : install php] *************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'p                                           hp7.4-fpm' is available"}
        to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP *********************************************************************
localhost                  : ok=47   changed=0    unreachable=0    failed=1

Error:   Something failed with the install - please inspect the logs above

dom-robinson, 2021-12-13:
HI @KE8AFF - there was an update to the installer that has switched php7.2 to php7.4.

I am commenting here to ask @Mahallon (who made the tweak) about that switch to better understand it.[https://github.com/jekhokie/raspberry-noaa-v2/commit/62cab79675e85599837e5c51f5a0b162b99145ad#diff-c0b4bd55d750aaa267f6b0f55402de6451ba69513694dc1f57abc72a640294ac](https://github.com/jekhokie/raspberry-noaa-v2/commit/62cab79675e85599837e5c51f5a0b162b99145ad#diff-c0b4bd55d750aaa267f6b0f55402de6451ba69513694dc1f57abc72a640294ac)

In the meanwhile if you edit ```/ansible/roles/webserver/tasks/main.yml``` lines 30-32 to be php7.2... i think you might get the install to run.

Crosswalkersam, 2021-12-13:
No, doesn't seem to work. If I do that, the install completes without error, but the webpanel is unreachable. Just "Error 502 - Bad gateway"

1062298577, 2021-12-24:
同样的问题  请问怎么解决

dom-robinson, 2021-12-30:
For ref this issue has arisen through introduction of bullseye support. Seems backward compatibility to buster install is lost. Try tagged release 1.8 (last known good on buster)

raufaser, 2022-01-14:
I worked around this issue by providing my system with a source of php7.4-fpm for armhf architecture.

I used the repo of Ondrej Sury.

```
sudo wget -O /etc/apt/trusted.gpg.d/php.gpg https://packages.sury.org/php/apt.gpg
echo "deb https://packages.sury.org/php/ buster main" | sudo tee /etc/apt/sources.list.d/php.list
sudo apt update
```

Have a nice one.

PR #491 (2022-01-25): Support buster and bullseye at once, drop stretch
It would be nice if someone could test this on bullseye, but I think it should work.

fixes #479
closed 2022-01-25 by dom-robinson