jekhokie/raspberry-noaa-v2 #480: [Error] Ballseye timesyncd not on os

thecivvie, 2021-12-15:
I got this error on a fresh install

TASK [common : disable systemd-timesyncd] **************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Could not find the requested service systemd-timesyncd: host"}

I have to edit ansible/roles/common/tasks/ntp.yml and remove the timsyncd to finish compiling

thecivvie, 2021-12-18:
I did another total fresh install and this error never occured. I had done an update to Bullseye from the cli so maybe something went wrong

ve3elb, 2021-12-18:
Have you tried using the prebuilt image. Give it a try. 

https://qsl.net/ve3elb/RaspiNOAA

thecivvie, 2021-12-20:
> Have you tried using the prebuilt image. Give it a try. https://qsl.net/ve3elb/RaspiNOAA
> […](#)

No, will do the next time I rebuild

closed 2021-12-20 by thecivvie

Lopastudio, 2024-02-22:
I solved it by creating a blank service called "systemd-timesyncd.service" and putting basically an empty service that doesnt do anything and it worked:
```
[Unit]
Description=My Service
After=network.target

[Service]
Type=simple
ExecStart=/usr/bin/python3 /opt/my_service/my_service.py
Restart=always

[Install]
WantedBy=multi-user.target

```