jekhokie/raspberry-noaa-v2 #484: Passes scheduling doesn't go to new Year

airlomba, 2021-12-30:
**Describe the bug**
My system is configured to schedule passes for 3 days. Everything works fine until this morning (30-12-2021) I find that the scheduler made predictions for 01-01-2022, as shown below, but the webpage passes doesn't show the passes for this days; only show for 30 and 31 December.

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Thu 30 Dec 09:43:20 UTC 2021
Current date/time (ms):  1640857400
Repo git hash:           30a07ff086ccd978e2dc1f962b3cc20330026687
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.1
Architecture:            Linux ADB-rpi9 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [45	Sat Jan  1 10:25:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220101-102510 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641032710 934 42 Southbound E
  * [32	Thu Dec 30 20:08:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20211230-200834 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640894914 938 32 Northbound W
  * [50	Sat Jan  1 18:21:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20220101-182143 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641061303 935 41 Northbound E
  * [27	Thu Dec 30 19:49:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20211230-194927 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640893767 927 54 Northbound W
  * [29	Thu Dec 30 22:15:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20211230-221538 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640902538 975 64 Northbound W
  * [38	Fri Dec 31 22:03:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20211231-220338 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640988218 978 82 Northbound W
  * [47	Sat Jan  1 08:17:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20220101-081748 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641025068 1000 81 Southbound E
  * [46	Sat Jan  1 21:51:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220101-215141 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641073901 986 79 Northbound E
  * [40	Fri Dec 31 19:56:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20211231-195631 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640980591 959 42 Northbound W
  * [49	Sat Jan  1 08:38:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20220101-083853 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641026333 947 41 Southbound W
  * [43	Sat Jan  1 07:38:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20220101-073851 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641022731 924 32 Southbound E
  * [37	Fri Dec 31 10:37:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20211231-103705 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640947025 937 54 Southbound E
  * [36	Fri Dec 31 19:24:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20211231-192408 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640978648 917 89 Northbound E
  * [41	Fri Dec 31 07:18:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20211231-071837 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640935117 933 37 Southbound E
  * [48	Sat Jan  1 19:44:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20220101-194432 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641066272 969 54 Northbound W
  * [44	Sat Jan  1 18:59:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20220101-185906 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1641063546 899 53 Northbound E
  * [28	Thu Dec 30 10:49:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20211230-104901 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640861341 940 68 Southbound E
  * [42	Fri Dec 31 18:41:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20211231-184122 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640976082 936 62 Northbound E
  * [35	Fri Dec 31 08:03:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20211231-080355 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640937835 960 56 Southbound E
  * [34	Thu Dec 30 19:01:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20211230-190110 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640890870 935 89 Northbound W
  * [39	Fri Dec 31 08:29:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20211231-082937 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640939377 987 80 Southbound W
  * [31	Thu Dec 30 18:28:00 2021 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20211230-182801 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1640888881 896 33 Northbound E
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: false
  noaa_15_gain: 40.2
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: false
  noaa_18_gain: 40.2
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: false
  noaa_19_gain: 40.2
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: 0
  meteor_m2_enable_bias_tee: false
  meteor_m2_gain: 40.2
  meteor_m2_sun_min_elevation: 6
  meteor_m2_sat_min_elevation: 30
  meteor_m2_memory_threshold: 600
---------------------------------------------
USB Device Map:
[... 49 lines cut ...]
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

**Server information (please complete the following information):**
 - OS:  Raspberry Pi OS (Linux rpi9 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux)
 - Python version: 3.9.2
 - SDR device: RTL-SDR v3

Best regards,
Emmanuel.
(CT7AFR)

colinluthier, 2022-01-28:
This was interesting. It certainly glitched on all installs on December 31st. It resolved the next day.

mihajlo2003petkovic, 2023-10-04:
Hmmm haven't really cgecked my website last New Year's Eve. This needs fixing, but you could mitigate it by settuig scheduling to 1 day in advance.