jekhokie/raspberry-noaa-v2 #485: Support for  shutdown button

Krolykke, 2021-12-31:
**Please Describe the problem to be solved**
The Raspberry Pi lacks a power button so one can not make it gracefully shutdown unless you have terminal access to it.

**(Optional): Suggest A Solution**

A solution can be found here:

https://www2.quartoknows.com/page/raspberry-pi-shutdown-button

With a little effort it could be modified to be configured through settings.yml

ve3elb, 2021-12-31:
Sorry I don’t really see the need for this. When you can have access to the pi via SSH, VNC or Teamviewer. So in this care this is not a problem to be solved.

colinluthier, 2022-01-28:
This would be an independent project to raspi noaa. It does not need to be included as it is entirely separate. If you need help implementing it on your pi text my cell phone. ;)

closed 2022-01-28 by colinluthier